pulp-platform/pulpino #179: PULPino inplace CM3 core in SYSTEM

thotaraghava, 2018-02-27:
HI all,

    I have replaced cortex-m3 with PULPino in my system which is under test.
    I have generated slm files for "helloworld" testcase.
    In my system, we have only one ROM in which both instruction and data should sit.
       So, I have put instruction data on top of the ROM and data just below that. 
   I am not using "peripherals" block, as we dont have such peripherals in our system.
      So, I am trying to drive boot_addr_i.

    @FrancescoConti  mentioned on other ticket that the given boot code is specific to pulpino stand alone platform, it needs to be changed for other cases.

    I have following questions.
       1. How this boot code should be changed in current scenario. Absolutely, its gray area for me.
       2. What should I drive boot_addr_i.

Thanks,

quangdaovu, 2018-02-27:
Why do you want to touch boot_code? Isn't `make helloworld.vsim` enough????

thotaraghava, 2018-02-27:
I am not simulating PULPino stand alone environment.
I just took PULPino RTL and replaced CM3 in my system.
I have taken instr data and data for "helloworld" testcase into my system.
In my system, we have one ROM for both instr and data, I just combined instr and data into one unit and pre-loaded my ROM.
I am not using instr RAM and data RAM from PULPino RTL.
In this case all instructions and data would come to PULPino from ROM which is in my system.

Now, please tell me. I dont think boot code in this case is necessary.
I am unable to see transactions at core_region.
Any clue.

quangdaovu, 2018-02-27:
1. You don't simulate PULPino alone but the test scenario is still the same. Just follow whatever provided testbench does. Software needs strict address space relationship. Your generated instruction and data will only work with instruction RAM and data RAM, if you messed up the address space then it won't work at all. In your case, you just print out instructions and data then copy paste them together to make your ROM then surely your software will NEVER work. 

If you don't understand how the system works then just follow the normal procedure, don't be too creative.

2. Clue: Read README about regenerate the bootcode, modify boot_code.c, generate new boot_code.sv and change MEM_LOAD method to STANDALONE.

FrancescoConti, 2018-02-27:
@quangdaovu when he says he took PULPino, he actually means mainly the RISCY core. He does not use a PULPino system, apparently.
I reiterate: the boot code depends on fundamental system-dependent parameters such as the memory map. Of course you can reuse part (maybe most) of the current boot code, but you'll have to adapt it significantly and we have no clue how (because we have no clue about your system).

Given that you're replacing an expensive ARM IP, I suppose you're working in an organization which most likely has documented the previous ARM-based system. You can use:
1. that documentation
2. the previous ARM boot code
3. the PULPino RISC-V boot code
to create a *new* boot code for your system. More than that, we cannot possibly help you.

thotaraghava, 2018-03-01:
Here, nobody can help with respect to PULPino except you.
I am the only one  working on it.

Yes, I am just using riscv_core as @FrancescoConti onti  mentioned.
@quangdaovu  said, generated instruction and data will work only with instruction RAM and data RAM in PULPIno. 
So kindly let me know, how can I generate instruction and data to work in my system.
I am also looking from my side to crack this.

Kindly reopen this ticket as I may need more information on this.

Thanks,

thotaraghava, 2018-03-01:
I can hack by using RISCY core + instr RAM + data RAM from pulpino platform, but our ultimate goal is to use OUR ROM for both instr and data.

thotaraghava, 2018-03-01:
Now, after reset,  core_instr_addr is pointing to 0x00000004.
Is it reset vector ??
I am storing instr+data in ROM from '0' location on wards.
If I can make above address pointing to 0x00000000, will this works.
This can be done by removing addition of 'd4 for "fetch_addr" in block riscv_prefetch_buffer.sv.
Please confirm.

FrancescoConti, 2018-03-01:
So to handle the problem on HW, you have to:
1. check that your instruction / data interfaces are appropriately connected to your memory (or your interconnect)
2. modify the boot address by driving the primary RISCY input `boot_addr_i` appropriately:
https://github.com/pulp-platform/riscv/blob/26875bef48c3d051ef414d6a0dfec0def8b7c2f3/riscv_core.sv#L65
e.g. if your reset vector will be located 0xbeef0000, you drive `boot_addr_i` with `32'hbeef0000`. This address depends on your system memory map. I don't think you need to remove anything in riscv_prefetch_buffer.sv to be honest, this is used to compute the next address for prefetching (necessary to make compressed instructions work).

That said, I'm not sure that I understand what is your problem - it seems to me that you already have these 2 points going. Do you see the wrong instruction being fetched/executed?

thotaraghava, 2018-03-01:
I am driving boot_addr_i with 0x00000000.
At present, I dont see any transactions around riscv_core.
core_instr_addr is pointing to 0x00000004.
No core_instr_req is coming out.

zarubaf, 2018-03-01:
`fetch_enable` is high?

thotaraghava, 2018-03-01:
I am not driving it. What it indicates.
Please suggest.

thotaraghava, 2018-03-01:
I tested by driving it 1'b1.
Suddenly, core_instr_req started coming.
Is it expected.

FrancescoConti, 2018-03-01:
Two things:
1. as the name suggests, you must drive `fetch_enable` to 1'b1 to enable the core to fetch instructions and hence start it.
2. in general, you must ***always*** drive all primary inputs to *something*. You cannot leave them undriven, or the core might enter an undetermined state.

thotaraghava, 2018-03-01:
ok. Sure.

thotaraghava, 2018-03-02:
Hi all,
    I have a quick question. You know, I am working on "Helloworld" testcase.
    If I had simulated in PULPino platform, will "Helloworld" is printed to show as testcase PASS. ?

    In my case, how can I make sure whether this testcase is PASSing or not.

Regards,

thotaraghava, 2018-03-02:
I need a confirmation regarding riscv_core protocol (req, gnt, rvalid ...).

I understand "rvalid" always follows "gnt" by exactly one cycle in both read and write transactions.

And "gnt" can come any time during req or after it.

please confirm.

zarubaf, 2018-03-02:
Pls read the RISCV [datasheet](http://www.pulp-platform.org/wp-content/uploads/2017/11/ri5cy_user_manual.pdf).

thotaraghava, 2018-03-02:
I read it, from that only I undertsand it.
It tells "rvalid" follows "gnt" by 1 cycle during read operation, but not clear information available for WRITE transaction.

FrancescoConti, 2018-03-02:
In this case I must say the datasheet is not too clear (or rather, takes a few things for granted).
I copy-paste and modify from the docs for [hwpe-stream](https://github.com/pulp-platform/hwpe-stream), which uses a very similar protocol. "Master" is the core in this context, "slave" is e.g. the memory.

Signal | Size | Description | Direction
-- | -- | -- | --
`req` | 1 bit | Handshake   request signal (1=asserted). | Master -> Slave
`gnt` | 1 bit | Handshake   grant signal (1=asserted). | Slave -> Master
`add` | 32 bit | Word-aligned   memory address. | Master -> Slave
`we` | 1 bit | Write enable   signal (1=write, 0=read). | Master -> Slave
`be` | 4 bit | Byte enable   signal (1=valid byte). | Master -> Slave
`wdata` | 32 bit | Data word to   be stored. | Master -> Slave
`rdata` | 32 bit | Loaded data   word. | Slave -> Master
`rvalid` | 1 bit | Valid loaded   data word (1=asserted). | Slave -> Master

`req` and `gnt` are used to *handshake* transactions between masters and slaves. 
Transactions are subject to the following rules: 
1.	A valid handshake occurs in the cycle when both `req` and `gnt` are asserted (i.e., transactions happen in the cycle when `req` and `gnt` are 1). This is true for both write and read transactions.
2.	The `rvalid` signal must be asserted in the cycle in which `rdata` is valid (for loads) or in the cycle in which the write transaction is concluded from the point of view of the master (for stores). For memories with one cycle of latency, this is always the cycle after the handshake happens.
3.	The assertion of `req` (transition 0->1) cannot depend combinationally on the state of `gnt`. On the other hand, the assertion of `gnt` (transition 0->1) can depend combinationally on the state of `req` (and typically it does). This rule avoids deadlocks in ping-pong logic.

thotaraghava, 2018-03-02:
Actually, I was reading datasheet which is in github.
@zarubaf  pointed datasheet which is very clear in all aspects.

zarubaf, 2018-03-02:
@FrancescoConti The datasheet is more than clear in all aspects. Even with timing diagrams.

closed 2018-03-02 by zarubaf

thotaraghava, 2018-03-02:
HI,

  Though I am driving boot_addr_i with 32'd0, after reset core_instr_addr is pointing to 32'h00000004 and 
   core_instr_req is asserted when core_instr_addr is 32'h00000080.

   And I do not see any core_lsu_req here.
   Is it expected for "helloworld" testcase.

Thanks

gautschimi, 2018-03-02:
if you don't know how to interface the processor core I strongly suggest 
to first look at the full pulpino system, simulate the "helloworld" 
there and then observe the interfaces, etc. this will give you much more 
insight than asking random questions and wasting our time.

if you don't want to do that have a look at the following three points:

1. verify that all inputs of the core driven by meaningful values

2. when the core requests an instruction, check if you give back a gnt, 
rvalid and the instruction. and if the core asks for more instructions 
after this.

3. maybe a screenshot of your interface would help

thotaraghava, 2018-03-02:
Hi,
  I am getting below error messages.

504474000000: Illegal instruction (core 0) at PC 0x000019ba:

Any clue for this.

Thanks,

thotaraghava, 2018-03-02:
I have copied instr and data to my one and only ROM. 
Above error message related to instr and data placement in ROM ?
What should I do.

thotaraghava, 2018-03-02:
I did not use scatter file while generating instr and data files in pulpino setup.
How can I use scatter file in this setup.

gautschimi, 2018-03-02:
how do you connect two interfaces to one ROM? and where does your data 
interface write to? you might want to use a WOM 
<http://repeater-builder.com/molotora/gontor/25120-bw.pdf> aswell.

no seriously, simulate pulpino as it is and first study the design 
before you go and modify anything. your knowledge is not big enough to 
do such simple modifications. you will never succeed if you don't invest 
time on your own and first study processors and how they work

thotaraghava, 2018-03-07:
I connect instr and data interfaces to ROM through bridge.
At any time, one interface is active.
I am sorry, for schedule point of view, I can not work on PULPino.

thotaraghava, 2018-03-08:
In my scenario, when I load all my instr and data into ROM, I am getting below error while simulating.

504474000000: Illegal instruction (core 0) at PC 0x000019ba:

Can anybody guess, where was the issue and how I should have dumped data in ROM.

And you can let me know, how scatterfile concept is used in instr and data file generation for helloworld testcase.

Thanks,

thotaraghava, 2018-03-09:
Could not see any comment on this yet. !!!

FrancescoConti, 2018-03-09:
> Could not see any comment on this yet. !!!

This is not the attitude that is going to get *any* answer *anywhere* on the internet.
You know, we have a job, and this job is not centered on answering questions on GitHub.

> In my scenario, when I load all my instr and data into ROM, I am getting below error while simulating.
504474000000: Illegal instruction (core 0) at PC 0x000019ba:
Can anybody guess, where was the issue and how I should have dumped data in ROM.

The key word here is *guess* as this is all we can do. What the simulator is telling is that @0x000019ba you have an illegal instruction.

thotaraghava, 2018-03-12:
Hi all,

     Could you please point me linker script used in this PULPino setup.
     I am also searching from my end.
   
thanks

FrancescoConti, 2018-03-15:
`crt0.riscv.S` is a boot script. `link.common.ld` is a linker script.

thotaraghava, 2018-03-16:
That's great. 
If we preload instr and data RAMs, I do not think, we need boot loader. Is it ?
When we change settings in linker script, will there be any change in data in l2_stim.slm and tcdm_bank0.txt ?

Thanks

thotaraghava, 2018-03-16:
Now, next step, I am going to do is removing instr and Data RAMs and load entire programme into one external ROM in my system.  In this case any change in boot and linker script.

thotaraghava, 2018-03-16:
I think, this question, I had asked earlier.
For "helloworld" testcase, what is the simulation output expecting?
should there be any print message of helloworld.

Please confirm.

Thanks

who93, 2018-03-17:
yes, there should be a message printed on UART - try to run it in ModelSim with make helloworld.vsim, there you see the output message

FrancescoConti, 2018-03-17:
The boot script sets up the minimal runtime stuff necessary to use C compiled code (basically, zeroes out a few data structures and creates the stack). It is therefore quite essential. Regarding any changes, in particular to the linker script, they depend on your system not on PULPino. For info on how to manage linker scripts, see here: https://sourceware.org/binutils/docs/ld/Scripts.html#Scripts

Regarding the various `stim.txt` files, they use indeed the `ADDRESS_DATA` (32 bit address, underscore, 32 bit data) format. Ignoring the address is not a good idea, as the data words are not necessarily contiguous...

thotaraghava, 2018-03-19:
Thanks @FrancescoConti  for your detailed answer above.
If ignoring address is not good idea in pre-load mode, could you please understand me how can I preload (l2_stim.slm, tcdm_bank0.slm).
All these questions coming because, I am not simulating in PULPino platform but in my system with ncSim.
My system = riscv_core + instr RAM + data RAM + system.

At present preloading the above memories with $readmemh after ignoring addresses. Simulation is not converging.

Thanks,

thotaraghava, 2018-03-22:
Hi all,

     For helloworld testcase, I have dumped instructions and data into RAMs.
     To my surprise, simulation is running for more time.
      In terminal, I could see message "setfile Done".
    Any clue why its been like that.

Thanks,

thotaraghava, 2018-03-22:
Is my pre-loading is okay ?
I suspect that.

Thanks,