pulp-platform/pulpino #181: Regarding new hardware accelerator

ravi02011993, 2018-03-07:
Hi all,
I have to add a new hardware accelerator to APB bus. So I made a dummy peripheral and added it along with all other peripherals that are connected to APB. when I give the command "make vcompile" it does not compile the new peripheral slave.  I think it is not detecting the peripheral. And when I try to simulate I got the same error.
Please help me how to solve the issue

Regards
Ravindranath

FrancescoConti, 2018-03-09:
Did you add the dummy peripheral in a new source file? If so, you will need to update https://github.com/pulp-platform/pulpino/blob/master/vsim/vcompile/build_rtl_sim.csh to make this visible from `make vcompile`.

BTW, see issue #142 for some people having similar needs.

ravi02011993, 2018-03-09:
Ok. Thank you

closed 2018-03-09 by ravi02011993