pulp-platform/pulpino #182: Potential forbidden axi write handshake dependency in debugging unit

barrydebruin, 2018-03-15:
Hi,

I added a new AXI slave peripheral to Pulpino, which basically waits for a message from a master and generates an awready if awvalid and wvalid are both active. I decided to test this peripheral by using the default testbench from _./tb/tb.sv_ in combination with _adv_dbg_if.axi4_write32(waddr, 1, wdata)_ call to emulate requests from a master device.

I ran into some issues that the adv_dbg_if.axi4_write32 call seems to wait for the awready flag after the awvalid is issued before it will continue to set the wvalid line. This implies that wvalid depends on awready, which it should not according to the standard. The code that seems to cause the dependency issue can be found on lines 519-545 of the _./ips/adv_dbg_if/rtl/adbg_axi_biu.sv_ file.

From the AMBA AXI and ACE Protocol Specification:
> Write transaction dependencies
>    - the master must not wait for the slave to assert AWREADY or WREADY before asserting AWVALID or WVALID
>    - the slave can wait for AWVALID or WVALID, or both before asserting AWREADY

url: [http://www.gstitt.ece.ufl.edu/courses/fall15/eel4720_5721/labs/refs/AXI4_specification.pdf](url)

FrancescoConti, 2018-03-15:
Hi @barrydebruin , thanks for the catch. Do you have a fix already for this? Otherwise we will for sure evaluate this internally.

barrydebruin, 2018-03-15:
I do not have a fix yet, other than rewriting the handshake of my slave device to be compliant with the rest of the pulpino slaves (assert AWREADY after AWVALID before WVALID is asserted).

This implies that this part of the protocol handshake is not implemented:

> - the slave can wait for AWVALID **or WVALID, or both** before asserting AWREADY

I tried some other things, such as asserting both AWVALID and WVALID manually on the master side, but the AXI interconnect will not let the WVALID signal through until the slave has pulled AWREADY high (which my slave did not do since it was waiting for both AWVALID and WVALID).

FrancescoConti, 2018-03-22:
We (should have) fixed the issue you mention in the master of https://github.com/pulp-platform/adv_dbg_if . I'm closing this issue, but feel free to reopen it and let us know if the problem is still there or hits some other module.
Thanks for pinpointing the problem!

closed 2018-03-22 by FrancescoConti

barrydebruin, 2018-03-23:
Hey,

Thank you for addressing the problem so quickly. However, I feel that the problem is still partially present in the AXI-interconnect logic.

It is easy to reproduce by letting a c-application send something to some AXI slave via memory mapped IO. I try to make the problem visuable in the following modelsim trace:
[image]

The blue and purple lines denote the write transaction that is initiated by the pulpino core (AXI-master). Both the _awvalid_ and _wvalid_ lines are pulled high at cursor _TimeA_. At time _TimeA + 1 (tick)_ the wvalid of the slave is pulled high. This delay is caused by the _AW_ADDR_DEC_ block, which depends on _awready_ (which is asserted by the slave). However, if we would have had a slave peripheral that doesn't set _awready_ before _wvalid_ is set, this situation would result in a deadlock (which is why I opened the issue initially).

Slightly more technical explanation:
1. _awvalid_ of master is set when the write in initiated
2. _awvalid_ is set on slave side after _AW_ADDR_DEC.push_DEST__o is asserted
3. _AW_ADDR_DEC.push_DEST_o_ is asserted when _AW_ADDR_DEC.awready_o & AW_ADDR_DEC.awvalid_i_ holds
4. _AW_ADDR_DEC.awready_o_ is not set if the slave is waiting for _awvalid_