pulp-platform/pulpino #185: Error while Running the helloworld.vsim

SumanthKK, 2018-03-19:
Hi,

I am trying to simulate the "helloworld.vsim" and I am getting an error of the permission denial as in the below snippet. I have cross checked the vsim path and the 64-bit installation, everything looks fine. I also tried running the vsim outside the pulpino to check if there is a permission problem for me, but I am able to run the vsim. 
I also ran the "helloworld.vsimc" in modelsim console, apparently there is an error with -c, not sure what the issue is though. I have looked into the previous issues with the helloworld, but nothing seems sort of mine. Please help to resolve the issue.
[image]
[image]

Thanks,
Sumanth.

who93, 2018-03-20:
have you tried the installation described in the last post in #167 ?

SumanthKK, 2018-03-20:
I tried it now, I have replaced "-64" in all the scripts. However, I am getting an error with "-do" option now which was mentioned in #167. Not sure if I need to replace the "-do" option as well. Any idea?

who93, 2018-03-20:
Had the same problem and switching to CentOS 6.8 solved it.

SumanthKK, 2018-03-20:
I am running on the university servers, I do not think we can change the OS.

SumanthKK, 2018-03-20:
@FrancescoConti Could please address this issue?

FrancescoConti, 2018-03-20:
What's the output of `make VERBOSE=1 helloworld.vsim` ?

SumanthKK, 2018-03-20:
Below is the output of the command "make VERBOSE=1 helloworld.vsim"
```
/usr/bin/cmake -H/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw -B/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 helloworld.vsim
make[1]: Entering directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
/usr/bin/cmake -H/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw -B/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build/CMakeFiles 3
make -f CMakeFiles/Makefile2 apps/helloworld/CMakeFiles/helloworld.vsim.dir/all
make[2]: Entering directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/depend
make[3]: Entering directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
cd /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build/CMakeFiles/crt0.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/build
make[3]: Entering directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make[3]: Nothing to be done for `CMakeFiles/crt0.dir/build'.
make[3]: Leaving directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
/usr/bin/cmake -E cmake_progress_report /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build/CMakeFiles 
[  0%] Built target crt0
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/depend
make[3]: Entering directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
cd /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/libs/Arduino_lib/core_libs /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build/libs/Arduino_lib/core_libs /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build/libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build
make[3]: Entering directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make[3]: Nothing to be done for `libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build'.
make[3]: Leaving directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
/usr/bin/cmake -E cmake_progress_report /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build/CMakeFiles  1
[ 33%] Built target Arduino_core
make -f libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/build.make libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/depend
make[3]: Entering directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
cd /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/libs/Arduino_lib/separate_libs /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build /net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build/libs/Arduino_li [... cut ...]
make[3]: Leaving directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
make[2]: Leaving directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make[1]: Leaving directory `/net/plato.ee.Virginia.edu/misan1/users/bengroup/workspace/Sumanth/Nextgen_processor/pulpino/sw/build'
make: *** [helloworld.vsim] Error 2
```

quangdaovu, 2018-03-20:
From your log, it seems like your modelsim/questasim path doesn't correct. 
How about the output of 'which vsim'?

who93, 2018-03-20:
maybe try a newer version (16.0.0.211 is working, but the latest one not).

FrancescoConti, 2018-03-20:
You probably just need to reconfigure cmake with `cmake_configure.riscv.gcc.sh`. as @quangdaovu pointed out, the environment variable which is supposed to contain the path of `vsim` does not contain anything. In the worst case, you can also hardwire the correct path  instead of `which vsim` inside of `cmake_configure.riscv.gcc.sh`

quangdaovu, 2018-03-20:
Press F to observer full waveform (or right-click and Zoom Full). You'r viewing just the very beginning of full waveform, therefore the core still in reset state.

SumanthKK, 2018-03-20:
I understand that. But my question is if,  I am expected to add the waveforms and run the simulations after executing the file "helloworld.vsim" or running the "helloworld.vsim" file would do the work for me and display the waveforms. 
I am just curious to know if there is anything wrong with my setup or it is the expected procedure. 

Thanks,
Sumanth.

who93, 2018-03-20:
in pulpino/vsim/waves you find some .tcl files for adding the waves, after adding them, just do run -all and you see the output

FrancescoConti, 2018-03-20:
Apart from the default waves pointed out by @who93 , you should add the waveforms you're actually interested in, so this is 100% the expected behavior.

closed 2018-03-20 by FrancescoConti

SumanthKK, 2018-03-20:
Cool! Thanks everyone for the help!