pulp-platform/pulpino #186: Memory Transaction not performing as expected

SumanthKK, 2018-03-24:
Hi @FrancescoConti,

As a part of taping out the Zero-riscy processor, we are developing our own SRAM and the interface to interact with the processor as per the memory transaction diagram mentioned in the manual. As per my understanding, the read or write transaction takes place when there is a request signal either from the processor or from AXI bus and with the write enable high for the write transaction. 

I am trying to cross-verify the developed SRAM with the memory model you have provided with the PULPino to check if the developed SRAM interface is functioning correctly. But, the memory transactions with your ram model are working completely different than expected. The data from the memory is being read out when there is a change in the address irrespective of the enable signal. The read value defaults to 32'b0 when there is a spike in the ram enable signal.  Similarly, the write value defaults to 32'b0, when the enable signal and the write enable signals are high and then the write value changes as address changes. Please refer the below-simulated waveforms of the memory transactions for details.
[image]

Also, I have no clue what the spikes are on the request, grant and enable signals and how a read operation takes place at the spike. Not sure if my understanding is wrong or problem with my setup or it is the expected behavior. Could please help me in resolving this issue? 

I would very much appreciate your help. 

Thanks,
Sumanth.

FrancescoConti, 2018-03-24:
Hi @SumanthKK ,

first of all a disclaimer: I know the guts of PULPino quite well, but I was not one of its designers (I have been much more involved in OpenPULP https://github.com/pulp-platform/pulp and PULPissimo https://github.com/pulp-platform/pulpissimo - the first has actually a longer history then PULPino, 5 years now :) ). So I don't know the reasons for some of the design choices in PULPino.

You can find the actual memory model that is being used here https://github.com/pulp-platform/pulpino/blob/master/rtl/components/sp_ram.sv . It's a quite simple one, EN is used only for writes (I don't know why this choice was made), WE is active positive. The actual memory model interface is `data_mem_*`, while the core master interface `core_data_*` uses the TCDM interface defined in the data sheet and discussed also in other issues.

The interpretation you gave above is misled by the presence of those spikes, which actually do not have any impact on the memory interface. Using your example, from left to right this is the interpretation I would give instead of yours:
1. `wdata` is driven to `0` from the core (it's not a default value!). In the second cycle, `rdata` is driven to the value of `mem[addr]` *before the write* (`X`)
2. `rdata` is `0` the cycle after the previous write is performed. This is because the memory model disregards the enable signal for reads, it only uses the `addr` (which stays constant).
3. as above: `rdata` changes when `addr` changes (with 1 cycle latency)
4. same as 3.

As to why the spikes happen: I don't honestly know, this may come from legitimate (combinational) logic, somewhat poorly designed combinational blocks, or a combination of the two. They have delta duration and they must not have any effects on the results (at least until you start using device models with some embedded delays - then they can cause problems).

SumanthKK, 2018-03-28:
Hi @FrancescoConti,

Thanks for the detailed explanation! I think it is not yet clear to me on the transaction. I will dig into more details and get back to you. Please keep this ticket open.

Thanks,
Sumanth.

closed 2019-05-03 by SumanthKK