pulp-platform/pulpino #187: printf - unsigned long long (64 bit)

GiuseppeDiGuglielmo, 2018-03-27:
I am using PULPino + RI$CY and I am aware that the core does not support 64bit arithmetic, but I assuming that it can be emulated thanks to the compiler.

To start with I compiled and run this simple snippet of code
```
#include <stdio.h>
int main(int argc, char** argv)
{
  unsigned long long var = 10;
  printf("INFO: 0x%llx\n", var);
  return 0;
}
```
and I noticed that it prints `INFO: 0xlx` and not the expected `INFO: 0xa`.

Am I either assuming or doing anything wrong? 

(I am on commit 7ad47f3761db74a8f681d952e125664fd5abe71e)

Thank you

GiuseppeDiGuglielmo, 2018-03-27:
I found the following solution at the moment. Do you have any comment?

```
#include <stdio.h>

#define SPLIT(a) (unsigned)(a>>32), (unsigned)a

int main(int argc, char** argv)
{
  unsigned long long var = 10;
  printf("INFO: 0x%x%x\n", SPLIT(var));
  return 0;
}
```

FrancescoConti, 2018-04-30:
Hi Giuseppe, I noticed that this issue was never answered. Indeed, the printf does not fully support unsigned long long, but your workaround should work fine.

closed 2018-04-30 by FrancescoConti