pulp-platform/pulpino #188: led_demo is working fine but no other program is giving desired output

dkum11, 2018-03-30:
Almost all other except led_demo doesn't run fine and shows like this:

""Device has been reset
Sending block addr 00000000 with 256 entries
Sending block addr 000003FC with 256 entries
Sending block addr 000007F8 with 256 entries
Sending block addr 00000BF4 with 256 entries
Sending block addr 00000FF0 with 256 entries
Sending block addr 000013EC with 256 entries
Sending block addr 000017E8 with 256 entries
Sending block addr 00001BE4 with 256 entries
Sending block addr 00001FE0 with 48 entries
Sending block addr 00100000 with 256 entries
Sending block addr 001003FC with 256 entries
Sending block addr 001007F8 with 256 entries
Sending block addr 00100BF4 with 256 entries
Sending block addr 00100FF0 with 256 entries
Sending block addr 001013EC with 146 entries
Starting device"

and LD0 starts glowing. 

Can someone help me in this?

gautschimi, 2018-03-31:
What program are you executing? it looks like you are missing the uart 
output. try to launch the program with a timeout as described in the 
readme. it could be that the program is too short and it is already over 
before any character is printed. the timeout keeps the connection to the 
uart open for a specific time, rather than closing it after completion.

dkum11, 2018-04-01:
Thanks, timeout was the issue.Now its working fine.

closed 2018-04-04 by dkum11