pulp-platform/pulpino #191: mret instruction causes a illegal instruction exception

MaddyR, 2018-04-04:
Hello,

While returing from an interupt , the end_exception routine restores back the registers of ri5cy core followed by a mret instruction

Ideally the mret instruction should restore the PC from MEPC, but even with mepc instruction set to the instruction  to be executed when intedrupt occurred i.e.  (pc+4), the mret instruction doesnt restore the PC and ends up into a illegal instr exception @0x84.

Anyone has the same problem,

jackfan00, 2018-04-25:
I had similar but different problem before, my problem is in crt0.riscv.S file, original end_except block return by "ret" instruction, so interrupt ISR return to error location, after I change to "mret" the problem is sovled.

MaddyR, 2018-04-29:
Solved,  we have an architecture based on risy core wherein we were using an older version of the riscy core against the newer version on pulpino.  Thus using the new core IP from the pulpino architecture solved our problem

closed 2018-04-29 by MaddyR