pulp-platform/pulpino #196: Unrecognized command line option '-m=32' with the pulp-riscv-gnu-toolchain

lstrz, 2018-04-25:
On a 64 bit Ubuntu I've compiled the toolchain as instructed in its repo:
 > ./configure --prefix=/opt/riscv --with-arch=rv32imc --with-cmodel=medlow --enable-multilib
> make

When preparing to compiling helloworld with
> ./cmake_configure.zeroriscy.gcc.sh
I get an error
> riscv32-unknown-elf-gcc: error: unrecognized command line option '-m=32'

Did I do something wrong?

paultoth, 2018-04-26:
Seeing the same issue when I run cmake_configure.riscv.gcc.sh using this toolchain. I did not have any issue with this when I used the toolchain from pulp-platform/ri5cy-gnu-toolchain.

  riscv32-unknown-elf-gcc: error: unrecognized command line option '-m32';

I'm running on Centos 7, fwiw.

Generalbrus, 2018-04-30:
Same issue here on OSX

FrancescoConti, 2018-05-01:
I think PULPino supports only the older toolchain found here at the moment: https://github.com/pulp-platform/ri5cy_gnu_toolchain ; we use newer toolchains on PULPissimo and OPENPULP; @haugoug can you confirm this?

@Generalbrus  note that OS X is not a supported target anyways

pweiss2, 2018-05-04:
I am running into the same issue with -m32. I tried to build the ri5cy_gnu_toolchain, but I get an error when I run "make", and now I am stuck...

cfns.gperf: In function ‘const char* libc_name_p(const char*, unsigned int)’:
cfns.gperf:101:1: error: ‘const char* libc_name_p(const char*, unsigned int)’ redeclared inline with ‘gnu_inline’ attribute
cfns.gperf:26:14: note: ‘const char* libc_name_p(const char*, unsigned int)’ previously declared here
cfns.gperf: At global scope:
cfns.gperf:26:14: warning: inline function ‘const char* libc_name_p(const char*, unsigned int)’ used but never defined
Makefile:1064: recipe for target 'cp/except.o' failed

lstrz, 2018-05-05:
I have had the same error and have fixed it since. I'll paste my instructions on how to compile pulpino on a fresh Ubuntu 18.04 install. It's quite a read, but has worked on two fresh installs for me. You might find some helpful steps there. All in all, do try to follow the instructions from the repo's README, but when you encounter an error, check whether my instructions help.

Dependencies:

1. sudo apt install cmake gawk gcc git gperf g++ libc6:i386 libgmp-dev libmpc-dev libmpfr-dev make perl python python-pip texinfo

2. pip install pyyaml

3. sudo ln -s /usr/bin/make /usr/bin/gmake

Toolchain:

4. git clone https://github.com/pulp-platform/ri5cy_gnu_toolchain.git

5. cd ri5cy_gnu_toolchain

6. Run make. It will download some files, encounter an error and stop.

7. cd build/src/newlib-gcc/gcc/cp

8. Open cfns.gperf in your favorite text editor and remove lines below the first comment (starting at line 19, inclusive) up until the line containing "%}" without the quotation marks. After that's done, right after the comment ends with "*/" without quotation marks, the next line should be "%}" without quotation marks.

9. gperf -o -C -E -k '1-6,$' -j1 -D -N libc_name_p -L C++ --output-file cfns.h cfns.gperf

10. Open except.c and on line 1043 add "Perfect_Hash::" without quotation marks exactly in front of "libc_name_p" without quotation marks.

11. cd back to ri5cy_gnu_toolchain and run make. It should not encounter any errors and should finish successfully.

Vivado:

12. Install Vivado 2015.1 (you just need Vivado, the SDK and the boards; default to /opt/Xilinx, used here too). Get a license for Vivado WebPACK 2015.1. It's free, requires registering. Google it. Alternatively, download the license from here (might not be available forever): https://goo.gl/f2YYkf

13. Add to the end of ~/.bashrc:

export XILINXD_LICENSE_FILE=<your path to the downloaded license>

export PATH=$PATH:/opt/Xilinx/Vivado/2015.1/bin

export PATH=$PATH:/opt/Xilinx/SDK/2015.1/gnu/arm/lin/bin

export PATH=$PATH:/opt/Xilinx/SDK/2015.1/bin

export PATH=$PATH:<path to where the binaries of the ri5cy toolchain reside, for example /opt/ri5cy_gnu_toolchain/install/bin>

14. source ~/.bashrc

This (re)loads the environment variables from .bashrc

Pulpino:

15. cd to somewhere where you'll place the pulpino repo.

16. git clone https://github.com/pulp-platform/pulpino.git

17. cd pulpino

18. ./update-ips.py

19. cd fpga

20. make all

21. cd sw/apps/spiload

22. make

23. cd ../../../../sw/

24. mkdir build

25. cd build

26. cp ../cmake_configure.riscv.gcc.sh .

27. ./cmake_configure.riscv.gcc.sh

28. make helloworld

Putting everything on an SD card:

29. Set up an SD card according to http://www.wiki.xilinx.com/Prepare+Boot+Medium

30. From pulpino/fpga/sw/sd_image copy BOOT.BIN, devicetree.dtb and uImate to the boot partition on the SD card. Extract rootfs.tar to the root partition on the SD card.

31. From pulpino/fpga/sw/apps/spiload copy spiload to the root partition on the SD card.

32. From pulpino/sw/build/apps/helloworld/slm_files copy spi_stim.txt to the root partition on the SD card.

Running:

33. Boot up the Zedboard with the SD card inside. If it doesn't boot into Linux automatically, make sure that all environment variable are at their default values (Google it).

34. Once booted up, login as root, cd to / and run ./spiload --timeout=2 ./spi_stim.txt. You should see "Hello World!!!!!" printed out.

If you want to compile a different instruction set for the pulpino platform, additional spets will need to be added before steps 9 and 20. Also, steps 26 and 27 will have to differ. Read the readmes in pulpino, pulpino/fpga, pulpino/sw and ri5cy_gnu_toolchain.

Generalbrus, 2018-05-05:
@pweiss2 I had the same issue and solved it by applying the changes of this PR https://github.com/pulp-platform/ri5cy_gnu_toolchain/pull/4

paultoth, 2018-05-07:
The comment from @pweiss2 and the two responses to that do not appear to be related to this issue. 
This issue is about trying to compile the pulpino tests using pulp-riscv-gnu-toolchain. 
@pweiss2 and followup responses appear to be compiling the ri5cy_gnu_toolchain and encountering some other problem.

FrancescoConti, 2018-05-08:
@paultoth you are right. To clarify: pulp-riscv-gnu-toolchain is not currently supported for PULPino.

closed 2018-05-08 by FrancescoConti

haresh-seenivasagan, 2024-03-15:
Hello  @FrancescoConti  is it still the case that  pulp-riscv-gnu-toolchain is not currently supported for PULPino <

FrancescoConti, 2024-03-15:
We do not really support PULPino any more. If you want an up-to-date minimal system for RISC-V cores, my recommendation is X-HEEP from EPFL: https://github.com/esl-epfl/x-heep which can be thought as  a "spiritual successor" to PULPino.
For a complete RISC-V micro-controller system, our PULPissimo https://github.com/pulp-platform/pulpissimo is the right place.

haresh-seenivasagan, 2024-03-15:
Thanks for your reply @FrancescoConti really appreciate it. I am student and i think it is a bit too late for me to switch my riscv processor. If you dont mind could you spare some time and have a look at this https://github.com/pulp-platform/pulpino/issues/408 . would really appreciate your advice