pulp-platform/pulpino #198: spi master gives output at all the four signals

ravi02011993, 2018-04-30:
[image]

Dear all,

Currently  I am working on SPI master to transfer the data into one particular address.
I have used the following commands for giving the output at sdo0.

spi_setup_dummy( 0 , 0);

  spi_setup_master(1); 
  spi_setup_cmd_addr(0x1b , 8 ,0x001050FF ,32);
  spi_set_datalen(120);
  spi_start_transaction(SPI_CMD_WR , SPI_CSN0);
spi_write_fifo( data , Datalen)

I have the following query

1. CS0 is only getting low and the other chip selects are not active. But the serial output is available
in all the four ports from sdo0- sdo3. Can someone tell me what is the reason for this?

Thanks in advance

Regards
Ravi

ravi02011993, 2018-05-02:
[image]

@FrancescoConti 
i have configured the apb_spi_master peripheral in spi_std mode. please find the attached screenshot.
spi_mode is 0 but the output is available in all the 4pins.

WaleedOsama, 2018-05-12:
Are you sure this is quad mode ? 

In quad mode data will be distributed on the four lines 

So if you send 4'b1101 in quad mode 

So sdo0 = 1 , sdo1 = 1 , sdo2 = 0 , sdo3 = 1

But in standard mode the 4 bits will be sent after each other on sd0.

But i noticed that in hardware of spi in standard mode they make the same output in sd0 on the other 3 lines , i do not know what is the use of such move .

but if you want standard mode , and you are seeing ( command , address , dummy , data ) on sdo0 then just disregard the other 3 output lines .

ravi02011993, 2018-05-12:
Yeah, I also thought to disregard the remaining outputs. Thank you