pulp-platform/pulpino #200: Branch_taken

LilachDery, 2018-04-30:
hi!
 i am working on PULPino . i am doing a research on branch predicting. i want to add ML based branch predictor to PULPino. for that, I want to add to trace_core_00_0.log the printing of bit 9 R/W BRANCH_TAKEN from Performance Counter Event Register (PCER),CSR Address: 0x7A0. i want to know each time there was a beq,bne,bge if the branch was taken or not. how can i do it? where do i enter the printing and how do i compile the new file and where do i insert the compiled file? thank you! my email is hadas@borger.co.il

FrancescoConti, 2018-04-30:
Closing as duplicate of #199

FrancescoConti, 2018-04-30:
You should be able to modify `riscv_tracer.sv` accordingly.

LilachDery, 2018-04-30:
thanks!
how do i specifically print the bit 9 R/W BRANCH_TAKEN from Performance Counter Event Register (PCER),CSR Address: 0x7A0. each time there is a beq/bne/bge command?

davideschiavone, 2018-05-02:
Hello, you should simply hack the tracer in the decode phase, basically you have to detect when such instruction comes, then print the signals coming from the PCER (in your case from the cs_register module)

best
Davide

closed 2018-06-27 by davideschiavone