pulp-platform/pulpino #201: About riscv_tracer

Liyang131313, 2018-05-04:
Hello, excuse me, 
I would like to ask what the "riscv_tracer" module is used for, and which stage of the processor pipeline does it work, how can I use it?
Can I use it to keep track of instructions and address information executed in parallel? I want to use it for the study of control flow integrity.
Thank you, best wish
A Chinese student

barrydebruin, 2018-05-28:
As far as I know, the riscv_tracer module can only be used in RTL simulation, as it contains several non-synthesizeable verilog constructs such as file IO.

The riscv_tracer will keep track of the instructions that are executed by the core while running an application.  By default, the 'make helloworld.vsim' example will print the execution trace to a 'trace_core_00.log' file. The first few lines look like this:
```console
    Time           Cycles       PC            Instr          Mnemonic
    18880000       455          00000080      00c0006f       jal x0, 12             
    18960000       457          0000008c      30501073       csrrw x0, x0, 0x305      
    19000000       458          00000090      00000093       addi x1, x0, 0            x1=00000000
    19040000       459          00000094      00008113       addi x2, x1, 0            x2=00000000  x1:00000000
    19080000       460          00000098      00008193       addi x3, x1, 0            x3=00000000  x1:00000000
    ...
```
Time is you simulation time step (25MHz in this case). Cycles corresponds to the relative cycle count. The first instruction takes two cycles, while the others take 1 each. PC is the program counter, which corresponds to the address offset in the .elf file. You can try to track the control flow by matching the PC from the trace file with the corresponding segment (i.e. function name) from the disassembled .elf file. 

However, I feel that this could kind of control-flow tracking could be done much quicker and just as accurate with a processor-simulator, instead of an RTL-simulation.

I hope this will help you a bit.

Liyang131313, 2018-05-29:
@barrydebruin 
Thank you very much for your patience and detailed solution.
I hope my design for control flow detection is hardware-level, so I need a synthesizeable hardware IP core instead of the processor-simulator,. It seems that riscv_tracer can only be used to assist my research. Is there a way to modify the riscv_tracer so that the module can be synthesized? By the way, is there a relevant tracking hardware IP or interface in the RISCV architecture?
Thanks again, Wish you a happy life！

barrydebruin, 2018-05-29:
There are some performance counters that can be read out after synthesis. However, these counters do not report as detailed as the trace. There are some C-functions available to set these counters up for a particular part of you application. This slideset[1] contains some examples (e.g. see slide 9).

The counters are defined in the following file:
https://github.com/pulp-platform/riscv/blob/master/riscv_cs_registers.sv#L694

I do not think I can help you with any advice on how to extend them. You should maybe try to get in contact with some of guys from ETH Zürich.

[1]: www.iis.ee.ethz.ch/~gmichi/asocd/exercises/ex_02.pdf

Liyang131313, 2018-06-04:
Excuse me, I'm sorry I don't have an email reminder, it's so slow to reply you. I can't open this slideset. I try studying for a while and then contact ETH Zürich to call for help..

FrancescoConti, 2018-06-04:
@Liyang131313 there is someone in ETH currently working on a trace module (as a master thesis project), but the quick answer is that no, the riscv_tracer cannot be synthesized and there is no trivial way to make it synthesizable.
As @barrydebruin suggested, performance counters might be enough for your use case -- that's already viable in HW.

Liyang131313, 2018-06-05:
@FrancescoConti 
What stage of the performance counter is in the pipeline?
Can I trouble you to help me get in touch with this master's degree in the ETC Research Tracking module? I want to rely on pulpino to build the hardware CFI, and I also read the paper to understand that ETC's Antonio Barresi and Thomas R. Gross are also studying control flow In the study of bending, I hope to contact them and discuss with them.
Thank you very much.