pulp-platform/pulpino #204: MiBench

Liyang131313, 2018-05-15:
Excuse me，Can I use MiBench to test ariane?
Thanks, wish all the best.

ellenFan, 2018-06-06:
I tried, but for some of the test cases (qsort for example) instrram will overflow

Liyang131313, 2018-06-06:
@ellenFan Will the inst ram overflow this situation in most cases? Could I share your migration benchmark? I have a variety of problems in the process of transplanting c-files into elf files, even though I have already built the pulpino project. Thank you, wish you a happy life.

ellenFan, 2018-06-06:
@Liyang131313 I just ran into the overflow problem. The basic math small inside automotive folder doesn't have an overflow issue but requires the floating point unit. I haven't tried others, but since the qsort_small already overflows, I assume that the qsort_large will also overflow. As to adding applications on pulpino, we could follow the instructions on the pulpino github page (I think it requires adding our application directory in CMakeLists.txt of the pulpino/sw/apps directory) and then rebuild it inside pulpino/sw/build directory.

I just found a closed issue related to overflow  https://github.com/pulp-platform/pulpino/issues/98. It was posted more than a year ago, hopefully, we could find our solution from that post.

Best,

Liyang131313, 2018-06-07:
@ellenFan Hi，How to run /pulpino/build/apps/bench on pulpino soc, elf files I build （used make all）have not been cross-compiled, how to set the environment variable?

Liyang131313, 2018-06-07:
@ellenFan  I found setting the environment variable using **cmake_configure.riscv.gcc.sh** under the build directory, but I have the following problem with make helloworld:

Using default architecture flags!!

-- GCC_MARCH= RV32I
-- USE_ZERO_RISCY= 
-- RISCY_RV32F= 
-- ZERO_RV32M= 
-- ZERO_RV32E= 
-- PL_NETLIST= 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lczhang/workspace/pulppinO/pulpino/build
Scanning dependencies of target bench
[  0%] Building C object libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o
/home/lczhang/workspace/pulppinO/pulpino/sw/libs/bench_lib/src/bench.c:1:0: error: bad value (RV32I) for -march= switch
 // Copyright 2017 ETH Zurich and University of Bologna.
 ^
make[2]: *** [libs/bench_lib/CMakeFiles/bench.dir/src/bench.c.o] Error 1
make[1]: *** [libs/bench_lib/CMakeFiles/bench.dir/all] Error 2
make: *** [all] Error 2
[lczhang@localhost helloworld]$

ellenFan, 2018-06-07:
@Liyang131313 Have you tried the riscv tool chain provided by the pulpino github? After I changed my original riscv tool chain to the one provided, the example projects could compile successfully.

Liyang131313, 2018-06-08:
@ellenFan Thank you, my problem has been solved. There are more than one set of tool chains in my environment and the tool chain is in conflict. I am now trying to use bench in the pulpino to do the test and have already run to get the waveform. Thank you again!

closed 2018-08-02 by zarubaf