pulp-platform/pulpino #209: axi_node module compile error

yeyh830101, 2018-05-30:
Hi Sir, 
I use vcs to compile the axi_node module.  There are some errors.  Logs like below:
Error - [SV-UIP] Unconnected interface port
../ips/axi_node/axi_node_wrap_with_slices.sv, 75
"axi_port_slave"
    The port 'axi_port_slave' of top-level module 'axi_node_wrap_with_slices' whose type is interface 'AXI_BUS' is left unconnected. It is illegal to leave the interface ports unconnected.
Please make sure that all the interface ports are connected.
Error - [SV-UIP] Unconnected interface port
../ips/axi_node/axi_node_wrap_with_slices.sv, 76
"axi_port_master"
    The port 'axi_port_master' of top-level module 'axi_node_wrap_with_slices' whose type is interface 'AXI_BUS' is left unconnected. It is illegal to leave the interface ports unconnected.
Please make sure that all the interface ports are connected.

FrancescoConti, 2018-05-30:
We do not have "official" support for VCS at the moment. I see two possible workarounds for the problem you see:
- degrade that error to a warning -- most simulators allow this, and in general this specific problem is not raised on QuestaSim (nor, as far as I know, in NCSim). So it shouldn't be a huge problem.
- check whatever module is instantiating `axi_node_wrap_with_slices` -- it will likely have an unconnected interface somewhere, you might be able to fix this from there.

Hope this helps.

closed 2018-06-05 by FrancescoConti

AlvinRolling, 2019-03-06:
It seemed this file "axi_node_wrap_with_slices.sv" is not used. So I just omitted from the compilation scripts, and there's no such errors.