pulp-platform/pulpino #210: Pulpino critical warnings and custom wrapper

aignacio, 2018-05-30:
Hello all,

I would like first to thanks to all team of ETH for the excellent project and contribution for the RISC-V project. I have been working in a project to use pulpino as my RISC-V processor for a cluster of tiles nodes and during the compilation steps, I'm porting the tcl's script to my target (xc7k325tffg900-2) but I realized that the scripts finished with synthesis right, but when I read the logs, I saw a lot of critical warnings such as:

> [Project 1-863] The design checkpoint file /home/anderson/projects/fpga/kc705_pulpino/ips_xilinx/xilinx_mem_8192x32/ip/xilinx_mem_8192x32.dcp was generated for a block design or an IP or BD by an out of context synthesis run and should not directly be used as a source in a Vivado flow to refer to an IP source.  As of 2017.1, the DCP from OOC runs will not contain XDC timing constraints because these are expected to be referred to by the IP .xci or .xcix file source. DCP files prior to 2017.1 will contain incorrect constraints because they were generated with default OOC clock period which will not likely match your top level clock constraints when used in the full design context. 
 
And as I'm using a FPGA that **has not a SoC embedded** I'm doing **my own top wrapper** that'll contain the PULPILO TOP, but I'm still confusing **where** I need to **instantiate** the **xilinx_mem_8192x32**  and the **xilinx_clock_manager**. Thanks!

FrancescoConti, 2018-06-05:
Hi @aignacio. The warnings are related to our flow which is probably not working correctly for newer versions of Vivado. I think we recommended Vivado 2015.2 originally, you will have to slightly change the flow if you want to use a newer version.

Regarding the two modules you mention, the BRAM is used within the memory blocks. The clock manager wraps the PLL to generate the clock, but probably you will need to take care of that in a different way given your different setup.

aignacio, 2018-06-05:
Thanks @FrancescoConti, I'm installing the old version to test it...

AlvinRolling, 2019-08-16:
Hi, @aignacio . I'm currently doing the same, implementing PULPino on FPGA without the ARM core. This issue should solve your problem. In a word, you can just read in the xci file instead of the dcp file, and there will be no warning. It will still work even if you decide not to change the scripts at all.