pulp-platform/pulpino #211: What is the max frequency the ri5cy processor can run?

MaddyR, 2018-06-01:
on ZC702 board the pulpino worked well at 50Mhz clk, but when the clock is raised to 200Mhz the results are unexpected, the core's computation are not in line with the same code tested at 50Mhz.

FrancescoConti, 2018-06-01:
I don't know precisely the maximum frequency you can reach, but I would not expect PULPino to be working beyond ~100 MHz, as the design is optimized for ASIC implementation, not FPGA.

closed 2018-06-01 by FrancescoConti

nvlinh87, 2019-01-28:
@FrancescoConti  How about target frequency for RI5CY standalone synthesis? I'm trying to synthesize it with 28nm lib.

FrancescoConti, 2019-01-28:
Here https://www.research-collection.ethz.ch/bitstream/handle/20.500.11850/314427/1/S3S_Quentin.pdf you can find some up-to-date results reached on a 22nm chip. Without applying body biasing, the SoC design achieves **670 MHz**, I am not sure however if the critical path is located within the core. I would assume that the target for 28nm shouldn't be too far away from that.

FrancescoConti, 2019-01-28:
The version of the core should be essentially the same configuration you are targeting, is that right @davideschiavone ?