pulp-platform/pulpino #214: toolchain install problem

yeyh830101, 2018-06-07:
Hi Sir,
    I'm trying to install ri5cy_gnu_toolchain-mater.zip.  Error happens during building. The log is below. Would you please help me ? Thanks a lot. 

./bfd.h:2511:66: note: in definition of macro ‘HOWTO’
   { (unsigned) C, R, S, B, P, BI, O, SF, NAME, INPLACE, MASKSRC, MASKDST, PC }
                                                                  ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/../include/opcode/riscv.h:106:3: error: initializer element is not constant
   (RV_X(x, 12, 20) << 12)
   ^
./bfd.h:2511:66: note: in definition of macro ‘HOWTO’
   { (unsigned) C, R, S, B, P, BI, O, SF, NAME, INPLACE, MASKSRC, MASKDST, PC }
                                                                  ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/elfxx-riscv.c:263:3: note: in expansion of macro ‘ENCODE_UTYPE_IMM’
   ENCODE_UTYPE_IMM (-1U) | ((bfd_vma) ENCODE_ITYPE_IMM (-1U) << 32),
   ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/../include/opcode/riscv.h:106:3: note: (near initialization for ‘howto_table[18].dst_mask’)
   (RV_X(x, 12, 20) << 12)
   ^
./bfd.h:2511:66: note: in definition of macro ‘HOWTO’
   { (unsigned) C, R, S, B, P, BI, O, SF, NAME, INPLACE, MASKSRC, MASKDST, PC }
                                                                  ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/elfxx-riscv.c:263:3: note: in expansion of macro ‘ENCODE_UTYPE_IMM’
   ENCODE_UTYPE_IMM (-1U) | ((bfd_vma) ENCODE_ITYPE_IMM (-1U) << 32),
   ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/elfxx-riscv.c:279:62: warning: left shift count >= width of type [-Wshift-count-overflow]
   ENCODE_UTYPE_IMM (-1U) | ((bfd_vma) ENCODE_ITYPE_IMM (-1U) << 32),
                                                              ^
./bfd.h:2511:66: note: in definition of macro ‘HOWTO’
   { (unsigned) C, R, S, B, P, BI, O, SF, NAME, INPLACE, MASKSRC, MASKDST, PC }
                                                                  ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/../include/opcode/riscv.h:106:3: error: initializer element is not constant
   (RV_X(x, 12, 20) << 12)
   ^
./bfd.h:2511:66: note: in definition of macro ‘HOWTO’
   { (unsigned) C, R, S, B, P, BI, O, SF, NAME, INPLACE, MASKSRC, MASKDST, PC }
                                                                  ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/elfxx-riscv.c:279:3: note: in expansion of macro ‘ENCODE_UTYPE_IMM’
   ENCODE_UTYPE_IMM (-1U) | ((bfd_vma) ENCODE_ITYPE_IMM (-1U) << 32),
   ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/../include/opcode/riscv.h:106:3: note: (near initialization for ‘howto_table[19].dst_mask’)
   (RV_X(x, 12, 20) << 12)
   ^
./bfd.h:2511:66: note: in definition of macro ‘HOWTO’
   { (unsigned) C, R, S, B, P, BI, O, SF, NAME, INPLACE, MASKSRC, MASKDST, PC }
                                                                  ^
/home/yyh/sw/ri5cy_gnu_toolchain-master/build/src/binutils/bfd/elfxx-riscv.c:279:3: note: in expansion of macro ‘ENCODE_UTYPE_IMM’
   ENCODE_UTYPE_IMM (-1U) | ((bfd_vma) ENCODE_ITYPE_IMM (-1U) << 32),
   ^
Makefile:1607: recipe for target 'elfxx-riscv.lo' failed
make[6]: *** [elfxx-riscv.lo] Error 1
make[6]: Leaving directory '/home/yyh/sw/ri5cy_gnu_toolchain-master/build/build-binutils-newlib/bfd'
Makefile:1649: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/yyh/sw/ri5cy_gnu_toolchain-master/build/build-binutils-newlib/bfd'
Makefile:1134: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/yyh/sw/ri5cy_gnu_toolchain-master/build/build-binutils-newlib/bfd'
Makefile:2599: recipe for target 'all-bfd' failed
make[3]: *** [all-bfd] Error 2
make[3]: Leaving directory '/home/yyh/sw/ri5cy_gnu_toolchain-master/build/build-binutils-newlib'
Makefile:831: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/yyh/sw/ri5cy_gnu_toolchain-master/build/build-binutils-newlib'
Makefile:194: recipe for target 'stamps/build-binutils-newlib' failed
make[1]: *** [stamps/build-binutils-newlib] Error 2
make[1]: Leaving directory '/home/yyh/sw/ri5cy_gnu_toolchain-master/build'
Makefile:28: recipe for target 'build' failed
make: *** [build] Error 2

prats2018, 2018-07-10:
Hi yeyh830101
  How could you resolve this error? I am getting the same error. I am running on Ubuntu 16.04 , 32-bit.

yeyh830101, 2018-07-10:
Use 64bit Ubuntu, the problem will be soloved.

prats2018, 2018-07-10:
Ok thank you.

Alessandro0110, 2018-08-01:
Hello, I have the same problem specified by the previous comment. I use Xubuntu 16.04 64bit. Can someone help me?

config.status: creating po/Makefile.in
config.status: executing depfiles commands
config.status: executing libtool commands
config.status: executing default-1 commands
config.status: executing bfd_stdint.h commands
config.status: executing default commands
make[5]: Nothing to be done for 'info'.
make[5]: Leaving directory '/ri5cy_gnu_toolchain-master/build/build-binutils-newlib/bfd/po'
make[5]: Entering directory '/ri5cy_gnu_toolchain-master/build/build-binutils-newlib/bfd'
make[5]: Nothing to be done for 'info-am'.
make[5]: Leaving directory '/ri5cy_gnu_toolchain-master/build/build-binutils-newlib/bfd'
Makefile:1649: recipe for target 'info-recursive' failed
make[4]: *** [info-recursive] Error 1
make[4]: Leaving directory '/ri5cy_gnu_toolchain-master/build/build-binutils-newlib/bfd'
Makefile:2599: recipe for target 'all-bfd' failed
make[3]: *** [all-bfd] Error 2
make[3]: Leaving directory '/ri5cy_gnu_toolchain-master/build/build-binutils-newlib'
Makefile:831: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/ri5cy_gnu_toolchain-master/build/build-binutils-newlib'
Makefile:194: recipe for target 'stamps/build-binutils-newlib' failed
make[1]: *** [stamps/build-binutils-newlib] Error 2
make[1]: Leaving directory '/ri5cy_gnu_toolchain-master/build'
Makefile:28: recipe for target 'build' failed
make: *** [build] Error 2

haugoug, 2018-08-02:
This toolchain is no more supported you should switch to the one here:
https://github.com/pulp-platform/pulp-riscv-gnu-toolchain.git

closed 2018-08-02 by haugoug

Alessandro0110, 2018-08-03:
I read in some posts that new toolchain doesn't support pulpino.

haugoug, 2018-08-03:
The toolchain supports the core used in pulpino but indeed there are probably some adaptations to do in the build process to make it work.

kangjian888, 2023-11-18:
> Hello, I have the same problem specified by the previous comment. I use Xubuntu 16.04 64bit. Can someone help me?
>
> ```
> config.status: creating po/Makefile.in
> config.status: executing depfiles [...]

Have anyone solve this problem?