pulp-platform/pulpino #227: Delay for the peripheral

ravi02011993, 2018-06-28:
Hi all,

I have defined one peripheral with two outputs and connected it to the APB bus. When I tried to write values into the signals by writing the c code on the core, I see there is a delay of about 9 cycles between the two signals to get written. I know from the core to the peripheral it takes about 3 cycles(core2axi and axi to apb). Then what is happening to the core during the remaining cycles?

[image]

closed 2018-07-02 by ravi02011993

jtxiao82, 2018-11-28:
Hi,

I will also interested in adding a custom hardware accelerator on pulpino. But there is not source code for the previous issue. Would you kindly share your source code or steps for adding a new peripheral?

Thanks!

ravi02011993, 2018-11-28:
Hi, I remember there is already some steps in some other issue. I dont remember the issue number. So, you can search again. If not, let me know. I will try to help you. Because it was long time i did that one. and I forgot it now.

jtxiao82, 2018-11-29:
Hi,

Thanks for your reply. I try to reference GPIO module and #70 for adding custom accelerator. The method will work for my current status

ravi02011993, 2018-11-29:
Exactly. I used the same one. All the best 👍