pulp-platform/pulpino #234: Understanding "spi_stim.txt"

dkum11, 2018-07-24:
Some lines in spi_stim.txt are as:

"00000000_00000013
 00000004_00000013
 00000008_00000013"

What does these 64 bits contains either "address","data","opcode" or "operand"?

barrydebruin, 2018-10-30:
Its the byte-address and the word that will be written onto that address in memory. Since everything is memory-mapped in pulpino, these values will end up instruction memory.

(see Fig. 2.1 in https://www.pulp-platform.org/docs/pulpino_datasheet.pdf for a memory-map)

I do not know the exact encoding  of the instruction words.

caifuxi, 2019-10-24:
To add to @barrydebruin 's answer

The encoding of the instruction words is actually the helloworld.bin file (I am assuming you are running that).

To understand it better, you can run "make helloworld.read" in /sw/build folder, and will get the helloworld.read file:

```
Disassembly of section .vectors:

00000000 <.vectors>:
   0:	00000013          	nop
   4:	00000013          	nop
   8:	00000013          	nop
   c:	00000013          	nop
  10:	00000013          	nop
  14:	00000013          	nop
  18:	00000013          	nop
  1c:	00000013          	nop
  20:	00000013          	nop
  24:	00000013          	nop
  28:	00000013          	nop
  2c:	00000013          	nop
  30:	00000013          	nop
  34:	00000013          	nop
  38:	00000013          	nop
  3c:	00000013          	nop
  40:	00000013          	nop
  44:	00000013          	nop
  48:	00000013          	nop
  4c:	00000013          	nop
  50:	00000013          	nop
  54:	00000013          	nop
  58:	00000013          	nop
  5c:	1040006f          	j	160 <ISR_I2C_ASM>
  60:	1180006f          	j	178 <ISR_UART_ASM>
  64:	12c0006f          	j	190 <ISR_GPIO_ASM>
  68:	1400006f          	j	1a8 <ISR_SPIM0_ASM>
  6c:	1540006f          	j	1c0 <ISR_SPIM1_ASM>
  70:	1800006f          	j	1f0 <ISR_TA_OVF_ASM>
  74:	1640006f          	j	1d8 <ISR_TA_CMP_ASM>
  78:	1a80006f          	j	220 <ISR_TB_OVF_ASM>
  7c:	18c0006f          	j	208 <ISR_TB_CMP_ASM>
  80:	00c0006f          	j	8c <_stext>
  84:	1b40006f          	j	238 <illegal_insn_handler>
  88:	1c80006f          	j	250 <ecall_insn_handler>
```
That's actually the instructions that will be sent through spi. You can compare the helloworld.read and spi_stim.txt to see contents