pulp-platform/pulpino #236: problem with cmake_configure.riscv.gcc.sh

Alessandro0110, 2018-08-02:
Hello, thanks for your work. I'm trying to execute the script cmake_configure.riscv.gcc.sh but I obtain the following result:

root@alessandro-VirtualBox:/usr/src/pulpino/sw/build# ./cmake_configure.riscv.gcc.sh 
-- The C compiler identification is unknown
-- The CXX compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/c++
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
CMake Error at CMakeLists.txt:20 (enable_language):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Check for working CXX compiler: /usr/bin/c++
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Detecting CXX compile features - done
-- Configuring incomplete, errors occurred!
See also "/usr/src/pulpino/sw/build/CMakeFiles/CMakeOutput.log".
See also "/usr/src/pulpino/sw/build/CMakeFiles/CMakeError.log".

I installed (I hope) correctly the toolchain indicated here https://github.com/pulp-platform/ri5cy_gnu_toolchain. I tried also with the new toolchain here but I obtained the same errors.

I don't know how to solve the problem.

Thank you in advance for the help.

FrancescoConti, 2018-08-02:
It seems the toolchain is not in your PATH when you launch CMake.

Alessandro0110, 2018-08-02:
I used this guide https://it.ccm.net/faq/393-bash-la-variabile-d-ambiente-path to insert in the PATH variable the install directory path. Is it right?

FrancescoConti, 2018-08-02:
`export PATH=/whatever/path/of/riscv/gcc/bin:$PATH`

closed 2018-08-02 by FrancescoConti

Alessandro0110, 2018-08-02:
Thanks I solved. Now I have an other problem I've tried to do make vcompile and I obtained this:

/bin/sh: 1: tcsh: not found
CMakeFiles/vcompile.dir/build.make:57: recipe for target 'CMakeFiles/vcompile' failed
make[3]: *** [CMakeFiles/vcompile] Error 127
CMakeFiles/Makefile2:173: recipe for target 'CMakeFiles/vcompile.dir/all' failed
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
CMakeFiles/Makefile2:180: recipe for target 'CMakeFiles/vcompile.dir/rule' failed
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
Makefile:168: recipe for target 'vcompile' failed
make: *** [vcompile] Error 2

I have executed the ./update-ips.py  without any errors.

zarubaf, 2018-08-02:
I am pretty sure the problem is exactly what the error is telling you:
```
/bin/sh: 1: tcsh: not found
```
Have you installed `tcsh`?

Alessandro0110, 2018-08-02:
In which step described in the readme has to be installed?

Alessandro0110, 2018-08-03:
Ok, thank you I solved that problem. Unfortunately I have an other problem that I described in #237, now I'm trying to solve it. 
Anyway thanks a lot.