pulp-platform/pulpino #237: doubt on "make helloworld.vsim"

Alessandro0110, 2018-08-03:
Hello thank you for your work. I have a doubt about the simulation of helloworld. When I try to run the simulation I obtain the same error as #110  so I try to use the same solution and the error dissapear. The problem now is the simulation because the clock isn't working as shown in the following picture:
[image]

I've also tried to substitute "which vsim" with the path of vsim but in that case I've obtained other errors.

Thanks for your help.

Alessandro0110, 2018-08-03:
If I don't use the solution proposed in #110 I obtain the following error in the modelsim console:
# source tcl_files/run.tcl
# vsim -quiet tb -L pulpino_lib -L axi_node_lib -L apb_node_lib -L axi_mem_if_DP_lib -L axi_spi_slave_lib -L axi_spi_master_lib -L apb_uart_sv_lib -L apb_gpio_lib -L apb_event_unit_lib -L apb_spi_master_lib -L fpu_lib -L apb_pulpino_lib -L apb_fll_if_lib -L core2axi_lib -L apb_timer_lib -L axi2apb_lib -L apb_i2c_lib -L zero_riscy_lib -L axi_slice_dc_lib -L riscv_lib -L apb_uart_lib -L axi_slice_lib -L adv_dbg_if_lib -L apb2per_lib "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+MEMLOAD=PRELOAD" -t ps -voptargs="+acc -suppress 2103" -GTEST="" -gRISCY_RV32F=0 -gZERO_RV32E=0 -gZERO_RV32M=0 -gUSE_ZERO_RISCY=0 
# Start time: 16:44:18 on Aug 03,2018
# ** Warning: (vsim-8311) System Verilog assertions are supported only in Questasim.
# ** Fatal: ** Fatal: (vsim-3828) Could not link 'vsim_auto_compile.so': cmd = '/home/alessandro/intelFPGA_pro/18.0/modelsim_ase/gcc-4.7.4-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/root@alessandro-VirtualBox_dpi_4001/linuxpe_gcc-4.7.4/vsim_auto_compile.so" /usr/src/pulpino/vsim/work/_dpi/auto_compile@/linuxpe_gcc-4.7.4/*.o   -Wl,-Bsymbolic '
# (vsim-50) A call to system(/home/alessandro/intelFPGA_pro/18.0/modelsim_ase/gcc-4.7.4-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/root@alessandro-VirtualBox_dpi_4001/linuxpe_gcc-4.7.4/vsim_auto_compile.so" /usr/src/pulpino/vsim/work/_dpi/auto_compile@/linuxpe_gcc-4.7.4/*.o   -Wl,-Bsymbolic  >'/tmp/questatmp.wILxnK' 2>&1) returned error code '1'.
# The logfile contains the following messages:
# /home/alessandro/intelFPGA_pro/18.0/modelsim_ase/gcc-4.7.4-linux/bin/../libexec/gcc/i686-pc-linux-gnu/4.7.4/ld: /usr/lib32/crti.o: unrecognized relocation (0x2b) in section `.init'
# /home/alessandro/intelFPGA_pro/18.0/modelsim_ase/gcc-4.7.4-linux/bin/../libexec/gcc/i686-pc-linux-gnu/4.7.4/ld: final link failed: Bad value
# collect2: error: ld returned 1 exit status
# 
# No such file or directory. (errno = ENOENT)
# 
# 
# FATAL ERROR while loading design
# Error loading design
# End time: 16:44:19 on Aug 03,2018, Elapsed time: 0:00:01
# Errors: 2, Warnings: 1
# 

I don't know how to solve the problem. Any suggestions?

Thank you for your cooperation in advance.

Alessandro0110, 2018-08-03:
I have tried to run some other applications like SPI_test,fibonacci etc but I obtained always the same waveforms, so I believe that there is actually a problem.

quangdaovu, 2018-08-04:
Try run your simulation longer.

Alessandro0110, 2018-08-04:
Here there is the picture of the simulation run for 10000ps:
[image]

quangdaovu, 2018-08-05:
The simulation starts from around 20000ns. Try "run -a".

Alessandro0110, 2018-08-10:
I solved the problem. I didn't read well the unit of measure of the clock. Thanks for the help.

closed 2018-08-14 by Alessandro0110