pulp-platform/pulpino #242: helloworld compile error

dragonware-com, 2018-09-04:
Using https://github.com/pulp-platform/ri5cy_gnu_toolchain, got this error:

/home/wsun/pulpino/sw/ref/crt0.riscv.S: Assembler messages:
/home/wsun/pulpino/sw/ref/crt0.riscv.S:248: Error: unrecognized opcode `mret'
CMakeFiles/crt0.dir/build.make:62: recipe for target 'CMakeFiles/crt0.dir/ref/crt0.riscv.S.o' failed
make[3]: *** [CMakeFiles/crt0.dir/ref/crt0.riscv.S.o] Error 1
CMakeFiles/Makefile2:232: recipe for target 'CMakeFiles/crt0.dir/all' failed
make[2]: *** [CMakeFiles/crt0.dir/all] Error 2
CMakeFiles/Makefile2:1832: recipe for target 'apps/helloworld/CMakeFiles/helloworld.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.dir/rule] Error 2
Makefile:701: recipe for target 'helloworld' failed
make: *** [helloworld] Error 2

dragonware-com, 2018-09-05:
I have to add that the ri5cy_gnu_toolchain is not built completely, it stopped with this issue:https://github.com/pulp-platform/ri5cy_gnu_toolchain/issues/27

nvlinh87, 2019-02-22:
Hi,
I got same message though my ri5cy_gnu_toolchain was compile OK.
ri5cy_gnu_toolchain/install/bin/riscv32-unknown-elf-gcc -o ./crt0.riscv.S.o -c crt0.riscv.S
crt0.riscv.S: Assembler messages:
crt0.riscv.S:248: Error: unrecognized opcode `mret'

I tried to comment out line 248 ("mret") then it run OK.
Have you found the root cause yet ? @dragonware-com

nvlinh87, 2019-02-26:
I just tried to compile ri5cy_gnu_toolchain again and the problem seems to be fixed.