pulp-platform/pulpino #244: Illegal instruction (core 0)

Alessandro0110, 2018-09-10:
Hello, thank you for your work. I'm trying to run a new application on Pulpino but I have some problems. First, I encountered the problem described in the topic #98. So, in the file core_region.sv, I substituted the value of INSTR_RAM_SIZE  with 131072. Then, in the file link.common.ld I substituted the sixth row with "instrram    : ORIGIN = 0x00000000, LENGTH = 0x20000". In this way the error disappeared. But when I run the simulation on modelsim, in the modelsim transcript, appears "Illegal instruction (core 0) at PC 0x00017a94:". This can be seen also in the following wave:

[image]

The same error appears many times but after the first time it presents always the following address "Illegal instruction (core 0) at PC 0x00016af4:".
Could I forgot something related to the procedure seen in the topic #98?

I ran the simulation for 100 us.

Thanks for your help.

Alessandro0110, 2018-09-10:
I see from the ".read" file of my application that at that address indicated in the error I have the following instructions:

00017a94 <uart_set_cfg>:
   17a94:	1a107737          	lui	a4,0x1a107
   17a98:	00470713          	addi	a4,a4,4 # 1a107004 <_stack_start+0x19fff004>
   17a9c:	00072603          	lw	a2,0(a4)
   17aa0:	1a1007b7          	lui	a5,0x1a100
   17aa4:	c0164633          	p.bset	a2,a2,0,1
   17aa8:	00c72023          	sw	a2,0(a4)
   17aac:	00478693          	addi	a3,a5,4 # 1a100004 <_stack_start+0x19ff8004>
   17ab0:	00c78513          	addi	a0,a5,12
   17ab4:	0085d813          	srli	a6,a1,0x8
   17ab8:	08300713          	li	a4,131 # 83 <__DYNAMIC+0x83>
   17abc:	0ff5f593          	andi	a1,a1,255
   17ac0:	00e52023          	sw	a4,0(a0)
   17ac4:	0106a023          	sw	a6,0(a3)
   17ac8:	0a700713          	li	a4,167 # a7 <_stext+0x1b>
   17acc:	00b7a42b          	p.sw	a1,8(a5!)
   17ad0:	00e7a023          	sw	a4,0(a5)
   17ad4:	00300793          	li	a5,3 # 3 <__DYNAMIC+0x3>
   17ad8:	00f52023          	sw	a5,0(a0)
   17adc:	0006a783          	lw	a5,0(a3)
   17ae0:	0f07f793          	andi	a5,a5,240
   17ae4:	c017c7b3          	p.bset	a5,a5,0,1
   17ae8:	00f6a023          	sw	a5,0(a3)
   17aec:	00008067          	ret

Seems that these instructions are related to the uart, so I added in the program and in its libraries the uart library but nothing changed. I continue to look for a solution.

Thanks for your help.

gautschimi, 2018-09-10:
You can check the memory content of the instr ram. My guess it was not loaded correctly. Could be a testbench issue or the script that is generating the stimuli file

Alessandro0110, 2018-09-11:
I check the content of the instr ram from the testbench and, as you said, the instructions are not loaded correctly. In particular the instruction ram is loaded correctly until it reaches the row 8191 that corresponds to a PC equal to 7FFC. Then the instruction ram is full of "x". In theory the instruction ram should be full of values until it reaches the row 28176 (PC = 1b840). So should I modify the script generating the stimuli file? Where is it?

Thanks for your help.

Alessandro0110, 2018-09-25:
Can anyone help me?

Thanks in advance

closed 2018-10-26 by Alessandro0110

Jaina-96, 2020-04-30:
@Alessandro0110 Can you post the solution to this issue?

I am using a custom SRAM of size 4K for the instruction memory and the data memory. When executing a simple HelloWorld program, Modelsim throws an error: "Illegal instruction (core 0) at PC 0x00000084:"

Thanks in advance.

prajwalkashyap, 2023-06-19:
@Alessandro0110 How did you fix this issue ? Could you share the steps ?