pulp-platform/pulpino #245: Error compiling FPGA

kllbrd, 2018-09-22:
hello, I'm taking this:

`INFO: [Opt 31-138] Pushed 0 inverter(s) to 0 load pin(s).
INFO: [Project 1-111] Unisim Transformation Summary:
  A total of 2 instances were transformed.
  IOBUF => IOBUF (IBUF, OBUFT): 2 instances

open_run: Time (s): cpu = 00:00:19 ; elapsed = 00:00:18 . Memory (MB): peak = 2196.660 ; gain = 489.363 ; free physical = 2401 ; free virtual = 5186
# write_hwdef -force -file ./pulpemu.hwdef
# source tcl/impl.tcl
## create_clock -period 50.000 -name clk      [get_nets {pulpino_wrap_i/clk}]
WARNING: [Vivado 12-507] No nets matched 'pulpino_wrap_i/spi_clk_i'.
## create_clock -period 40.000 -name spi_sck  [get_nets {pulpino_wrap_i/spi_clk_i}]
ERROR: [Vivado 12-1387] No valid object(s) found for create_clock constraint with option '-objects [get_nets pulpino_wrap_i/spi_clk_i]'.
Resolution: Check if the specified object(s) exists in the current design. If it does, ensure that the correct design hierarchy was specified for the object. If you are working with clocks, make sure create_clock was used to create the clock object before it is referenced.

    while executing
"source tcl/impl.tcl"
    (file "tcl/run.tcl" line 105)
INFO: [Common 17-206] Exiting Vivado at Sun Sep 23 02:32:53 2018...
Makefile:6: recipe for target 'synth' failed
make[1]: *** [synth] Error 1
make[1]: Leaving directory '/run/media/kllbrd/work/PULP/pulpino/fpga/pulpemu'
Makefile:46: recipe for target 'pulpemu/pulpemu_top.bit' failed
make: *** [pulpemu/pulpemu_top.bit] Error 2`

and the pulpino.vds logs:

`pulpino/pulpino.runs/synth_1/pulpino.vds:CRITICAL WARNING: [Synth 8-3321] create_clock attempting to set clock on an unknown port/pin for constraint at line 2 of /run/media/kllbrd/work/PULP/pulpino/fpga/pulpino/constraints.xdc. [/run/media/kllbrd/work/PULP/pulpino/fpga/pulpino/constraints.xdc:2]
Binary file pulpino/pulpino.runs/synth_1/vivado.pb matches
pulpino/pulpino.runs/synth_1/runme.log:CRITICAL WARNING: [Synth 8-3321] create_clock attempting to set clock on an unknown port/pin for constraint at line 2 of /run/media/kllbrd/work/PULP/pulpino/fpga/pulpino/constraints.xdc. [/run/media/kllbrd/work/PULP/pulpino/fpga/pulpino/constraints.xdc:2]`

Any whats is going on?
Thanks

kllbrd, 2018-09-24:
Solved this, but another problem is that spiload running the spi_stim.txt generated by helloworld app with no "helloworld" printing. Only a led on zedboard turns red...
Any idea??

Thanks again!

closed 2018-09-24 by kllbrd

zzlzzlwithzzl, 2021-01-13:
Hey, have you solved the problem?