pulp-platform/pulpino #252: SPI flash external memories

aignacio, 2018-10-22:
Hello guys from PULP,

Does you have a flash driver to use with openocd to program external flash memories through the SPI master peripheral with pulpino? (flash bank test ....)

barrydebruin, 2018-12-11:
There is a test program that will program an SPI flash via the SPI master device. It's called testSPIMaster and can be found in pulpino/sw/apps/imperio/testSPIMaster. It works with the SPI flash module from the Pulpissimo repo: https://github.com/pulp-platform/pulpissimo/tree/master/rtl/vip/spi_flash

kingweio, 2019-05-25:
@barrydebruin Hi， I have tried according to your advice,but an error "RX string: ERROR: Spansion SPI flash not found", how can I deal with it?