pulp-platform/pulpino #256: bug of riscv_tracer module???

IC-Dream, 2018-11-14:
Hi, 

I wrote a simple assembly case like the below:
[image]

However, the trace log would print some X state, which is not expected.
[image]

What's more, Does the multi-cycles instructions of FP stall the  integer pipeline ???

Thanks 
Dream

IC-Dream, 2018-11-14:
it seems that the lui committed ahead of fmadd.s in gray area from waveform
[image]

IC-Dream, 2018-11-14:
Also, for APU two-cycles operations, like the fadd.s, the riscv_tracer also can't handle it, since they update FPR with write port W1. Thanks.

Liyang131313, 2018-12-06:
@IC-Dream I would like to ask you in what software to view the waveform, do you know how to install ModelSim in CentOS7?

IC-Dream, 2018-12-07:
@Liyang131313 use Verdi to view the waveform, i have no experience to install ModeSim.

Liyang131313, 2018-12-07:
@IC-Dream Me too, but how do you run the benchmarks that come with the pulpino platform, for example，aes_cbc，conv2d，dhrystone，ecc ，fir and so on.

IC-Dream, 2018-12-07:
@Liyang131313 Use VCS to run simulation.

Liyang131313, 2018-12-07:
@IC-Dream I am running "make all" under the file named soc_env, there is a call to VCS, it seems that only one .c file can be run. And those benchmarks contain multiple .c files.