pulp-platform/pulpino #260: use the performance counter

Liyang131313, 2018-12-06:
How to use the performance counter to observe the running of programs in the PULPino repository

barrydebruin, 2018-12-07:
There are some examples in the sequential_tests directory. See for example: https://github.com/pulp-platform/pulpino/blob/master/sw/apps/sequential_tests/convolution/convolution.c

Liyang131313, 2018-12-08:
sorry，The link you sent does not describe how to use performance counters.@barrydebruin

barrydebruin, 2018-12-08:
I think it does. Basically you have to wrap these calls around the program you want to profile:

```
  perf_reset();
  perf_enable_id(EVENT_ID,1);

  Conv3x3_Scalar(In, Out, IMG_ROW, IMG_COL, Kernel3x3_Scalar);

  perf_stop();
  perf_print_all();
```

As is also pointed out in slide 9 of this tutorial: https://iis-people.ee.ethz.ch/~gmichi/asocd/exercises/ex_02.pdf

Liyang131313, 2018-12-16:
Do you need to add a function library or header file?@barrydebruin
[des.zip](https://github.com/pulp-platform/pulpino/files/2683347/des.zip)

Could you please help me add it?

Liyang131313, 2018-12-16:
[image]
[image]
[image]

Can the final performance count result be read directly?

What is the difference between PCCR_in[10:0], PCCR_inc[10:0] and PCCR_inc_q[10:0]?Why PCCR is not 32 32-bit registers？

@barrydebruin @zarubaf @Razer6 @wallento

barrydebruin, 2018-12-16:
> Do you need to add a function library or header file?@barrydebruin
> [des.zip](https://github.com/pulp-platform/pulpino/files/2683347/des.zip)
> 
> Could you please help me add it?

The functionality of these performance counter functions is described in 
- sw/libs/bench_lib/src/bench.c 
- sw/libs/bench_lib/inc/bench.h

Compiling bench_lib as a static library (as described in sw/libs/bench_lib/CmakeLists.txt) and linking it together with your application sources (including the headers in sw/libs/bench_lib/inc; see the add_application macro in sw/apps/CMakeLists.txt) will make it work.

Liyang131313, 2018-12-17:
Thank you very much for your patience.
But in my second picture, when I get the signals, I don't see PCCR_n and PCCR_q. What's going on? @barrydebruin

Liyang131313, 2018-12-21:
Excuse me，
when I get the signals, I don't see PCCR_n and PCCR_q. What's going on? @zarubaf @Razer6 @wallento

pallaviboreddy, 2020-01-27:
hi,
the bench application codes like fft, fir present in pulpino/sw/apps/bench is not getting executed.
what might be the steps to be considered.i have done it as,
1../cmake_configure.riscvfloatg.gcc.sh
2.make fir
3.make vcompile
4.make fir.vsim
please suggest