pulp-platform/pulpino #263: Can’t send over Uart

yehiafarouk, 2018-12-09:
I am trying to send over uart. Without changing anything.
I tried the testUART app exists and i tried my own
But i can’t find results 
There is no transition on tx and rx it is only valued by one

barrydebruin, 2018-12-09:
The testUART app assumes a loopback i.e. it tries to send a sequence over tx and then expects to read it back on the rx pin.

You have to slightly modify the testbench (tb.sv) to make this test pass. The following part:
```
  pulpino_top
  #(
    .USE_ZERO_RISCY    ( USE_ZERO_RISCY ),
    .RISCY_RV32F       ( RISCY_RV32F    ),
    .ZERO_RV32M        ( ZERO_RV32M     ),
    .ZERO_RV32E        ( ZERO_RV32E     )
   )
  top_i
  (
    .clk               ( s_clk        ),
    .rst_n             ( s_rst_n      ),
...
    .uart_tx           ( uart_rx      ),
    .uart_rx           ( uart_tx      ),
...
```

Should be modified as follows:
```
  pulpino_top
  #(
    .USE_ZERO_RISCY    ( USE_ZERO_RISCY ),
    .RISCY_RV32F       ( RISCY_RV32F    ),
    .ZERO_RV32M        ( ZERO_RV32M     ),
    .ZERO_RV32E        ( ZERO_RV32E     )
   )
  top_i
  (
    .clk               ( s_clk        ),
    .rst_n             ( s_rst_n      ),
...
    .uart_tx           ( uart_rx      ),
    .uart_rx           ( uart_rx      ), // <- loop back uart_tx to uart_rx
...
```

barrydebruin, 2018-12-09:
As a sidenote, the standard uart_bus module in tb.sv does not work with different clock frequencies (only the default 25MHz works out of the box). However, by modifying the following lines it will also work with other frequencies:

In tb.sv change:

```
  uart_bus
  #(
    .BAUD_RATE(BAUDRATE),
    .PARITY_EN(0)
  )
  uart
  (
    .rx         ( uart_rx ),
    .tx         ( uart_tx ),
    .rx_en      ( 1'b1    )
  );
```

to

```
  uart_bus
  #(
    .BAUD_RATE(BAUDRATE),
    .PARITY_EN(0),
    .CLK_PERIOD(`CLK_PERIOD)
  )
  uart
  (
    .rx         ( uart_rx ),
    .tx         ( uart_tx ),
    .rx_en      ( 1'b1    )
  );
```

And in tb/uart.sv change:

```
interface uart_bus
  #(
    parameter BAUD_RATE = 115200,
    parameter PARITY_EN = 0
    )
  (
    input  logic rx,
    output logic tx,

    input  logic rx_en
  );
  timeunit      1ps;
  timeprecision 1ps;

  localparam BIT_PERIOD = (1000000000/BAUD_RATE*1000);
```
to
```
interface uart_bus
  #(
    parameter BAUD_RATE = 115200,
    parameter PARITY_EN = 0,
    parameter CLK_PERIOD = 40
    )
  (
    input  logic rx,
    output logic tx,

    input  logic rx_en
  );
  timeunit      1ps;
  timeprecision 1ps;

  localparam BIT_PERIOD = (1000000000/BAUD_RATE*1000/40*CLK_PERIOD);
```

barrydebruin, 2018-12-11:
The UART Tx line should be valued 1 in idle mode, so that seems ok to me. I do not know about your exact tool and build configuration. The standard build flow is something like this:

- git clone https://github.com/pulp-platform/pulpino.git
- ./update_ips.sh
- cd sw; mkdir build; cd build; cp ../cmake_configure.riscv.gcc.sh .; make vcompile -j8; make -j8
- make helloworld.vsimc
- "Hello World!" will be printed in console

Did you deviate from any of this?

wooolf0455, 2018-12-12:
I do it like what your said ,but I can not get "helloworld " in console,I'm looking for your advice @barrydebruin #264