pulp-platform/pulpino #264: Modelsim error:error loading design

wooolf0455, 2018-12-12:
I just want to run helloworld test, I chose cmake_configure.riscv.gcc.sh , and the steps what I do is as same as what README.md told me. The error happened at the last step:
anyone has the same question?or someone  help me ,thanks a lot!!!
ps:I'm new oneo(╥﹏╥)o

wooolf0455, 2018-12-12:
The contents of modelsim transcription are as follows ： 
source tcl_files/run.tcl
# vsim -do "source tcl_files/run.tcl;" 
# Start time: 10:37:33 on Dec 12,2018
# ** Note: (vsim-3812) Design is being optimized...
# ** Error: /export/homeO5/yzhou/Documents/pulpino/vsim/..//tb/tb.sv(388): Can't have packed array of integer type.
# ** Error: /export/homeO5/yzhou/Documents/pulpino/vsim/..//tb/tb.sv(388): Vopt Compiler exiting
# ** Error: (vopt-2064) Compiler back-end code generation process terminated with code 211.
# ** Note: (vopt-143) Recognized 1 FSM in architecture body "uart_receiver(rtl)".
...
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_BW_allocator(fast)".
# Error loading design

wooolf0455, 2018-12-12:
[yzhou@cloud017 ~/Documents/pulpino/sw/build-riscv]$make VERBOSE=1 helloworld.vsimc
/export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -H/export/homeO5/yzhou/Documents/pulpino/sw -B/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 helloworld.vsimc
make[1]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
/export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -H/export/homeO5/yzhou/Documents/pulpino/sw -B/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv --check-build-system CMakeFiles/Makefile.cmake 0
/export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_progress_start /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/CMakeFiles 4
make -f CMakeFiles/Makefile2 apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all
make[2]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
cd /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino/sw /export/homeO5/yzhou/Documents/pulpino/sw/libs/bench_lib /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/libs/bench_lib /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/libs/bench_lib/CMakeFiles/bench.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make[3]: Nothing to be done for `libs/bench_lib/CMakeFiles/bench.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
[  0%] Built target bench
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
cd /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino/sw /export/homeO5/yzhou/Documents/pulpino/sw /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/CMakeFiles/crt0.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make[3]: Nothing to be done for `CMakeFiles/crt0.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
[  0%] Built target crt0
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
cd /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino/sw /export/homeO5/yzhou/Documents/pulpino/sw/libs/string_lib /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/libs/string_lib /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/libs/string_lib/CMakeFiles/string.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make[3]: Nothing to be done for `libs/string_lib/CMakeFiles/string.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
[  0%] Built target string
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
cd /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino/sw /export/homeO5/yzhou/Documents/pulpino/sw/libs/sys_lib /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/libs/sys_lib /export/homeO5/yzhou/Documents/pulpino/sw/build-riscv/libs/sys_lib/CMakeFiles/sys.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make[3]: Nothing to be done for `libs/sys_lib/CMakeFiles/sys.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
[ 25%] Built target sys
[... cut ...]
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc] Error 1
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all] Error 2
make[2]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule] Error 2
make[1]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino/sw/build-riscv'
make: *** [helloworld.vsimc] Error 2

wooolf0455, 2018-12-12:
[yzhou@cloud017 ~/Documents/pulpino2/sw/build]$make VERBOSE=1 helloworld.vsimc
/export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -H/export/homeO5/yzhou/Documents/pulpino2/sw -B/export/homeO5/yzhou/Documents/pulpino2/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 helloworld.vsimc
make[1]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
/export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -H/export/homeO5/yzhou/Documents/pulpino2/sw -B/export/homeO5/yzhou/Documents/pulpino2/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
/export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_progress_start /export/homeO5/yzhou/Documents/pulpino2/sw/build/CMakeFiles 4
make -f CMakeFiles/Makefile2 apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all
make[2]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
cd /export/homeO5/yzhou/Documents/pulpino2/sw/build && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino2/sw /export/homeO5/yzhou/Documents/pulpino2/sw/libs/bench_lib /export/homeO5/yzhou/Documents/pulpino2/sw/build /export/homeO5/yzhou/Documents/pulpino2/sw/build/libs/bench_lib /export/homeO5/yzhou/Documents/pulpino2/sw/build/libs/bench_lib/CMakeFiles/bench.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make[3]: Nothing to be done for `libs/bench_lib/CMakeFiles/bench.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
[  0%] Built target bench
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
cd /export/homeO5/yzhou/Documents/pulpino2/sw/build && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino2/sw /export/homeO5/yzhou/Documents/pulpino2/sw /export/homeO5/yzhou/Documents/pulpino2/sw/build /export/homeO5/yzhou/Documents/pulpino2/sw/build /export/homeO5/yzhou/Documents/pulpino2/sw/build/CMakeFiles/crt0.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make[3]: Nothing to be done for `CMakeFiles/crt0.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
[  0%] Built target crt0
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
cd /export/homeO5/yzhou/Documents/pulpino2/sw/build && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino2/sw /export/homeO5/yzhou/Documents/pulpino2/sw/libs/string_lib /export/homeO5/yzhou/Documents/pulpino2/sw/build /export/homeO5/yzhou/Documents/pulpino2/sw/build/libs/string_lib /export/homeO5/yzhou/Documents/pulpino2/sw/build/libs/string_lib/CMakeFiles/string.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make[3]: Nothing to be done for `libs/string_lib/CMakeFiles/string.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
[  0%] Built target string
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
cd /export/homeO5/yzhou/Documents/pulpino2/sw/build && /export/homeO5/yzhou/Documents/cmake/cmake-3.11.4-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /export/homeO5/yzhou/Documents/pulpino2/sw /export/homeO5/yzhou/Documents/pulpino2/sw/libs/sys_lib /export/homeO5/yzhou/Documents/pulpino2/sw/build /export/homeO5/yzhou/Documents/pulpino2/sw/build/libs/sys_lib /export/homeO5/yzhou/Documents/pulpino2/sw/build/libs/sys_lib/CMakeFiles/sys.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/build
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make[3]: Nothing to be done for `libs/sys_lib/CMakeFiles/sys.dir/build'.
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
[ 25%] Built target sys
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/depend
make[3]: Entering directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
[... cut ...]
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc] Error 12
make[3]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all] Error 2
make[2]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule] Error 2
make[1]: Leaving directory `/export/homeO5/yzhou/Documents/pulpino2/sw/build'
make: *** [helloworld.vsimc] Error 2

this is the result after I added VERBOSE=1 . maybe it's more clear

quangdaovu, 2018-12-12:
> ** Error: /export/homeO5/yzhou/Documents/pulpino/vsim/..//tb/tb.sv(388): Can't have packed array of integer type.

Fix your testbench.

wooolf0455, 2018-12-12:
@quangdaovu the code I did not find packed array of integer type.  what is more ,line 388 is the end of the code. I think it may be not associated to the code.

wooolf0455, 2019-02-26:
I found it!!!! at pulpino-1/tb/mem_dpi.svh,
            buffer[i]   = rdata_temp_arr[j][ 7:0];
            buffer[i+1] = rdata_temp_arr[j][15:8];
            buffer[i+2] = rdata_temp_arr[j][23:16];
            buffer[i+3] = rdata_temp_arr[j][31:24];
error happened here.
The following is what I modified:
            automatic logic [31:0] xxx_tmp=rdata_temp_arr[j];
            buffer[i]= xxx_tmp[7:0];
            buffer[i+1]=xxx_tmp[15:8]; 
            buffer[i+2]=xxx_tmp[23:16];
            buffer[i+3]=xxx_tmp[31:24]; 
            //buffer[i]   = rdata_temp_arr[j][ 7:0];
            //buffer[i+1] = rdata_temp_arr[j][15:8];
            //buffer[i+2] = rdata_temp_arr[j][23:16];
            //buffer[i+3] = rdata_temp_arr[j][31:24];
and then,there is "helloworld" in console.But I don't know the reason why the error happened.
If you know the reason,tell me please.Thanks!

closed 2019-02-26 by wooolf0455