pulp-platform/pulpino #267: Use the Pmod ports of the Zyboard

sofchet, 2018-12-16:
Hi,
I implemented the Pulpino on the zyboard ( I did run gpio_demo successfully), now I want to read a digital signal over the Pmod ports of the zyboard.  do you know how am I supposed to do this ? 
`#define BTN_R_BIT   (1 << 19)
#define BTN_U_BIT   (1 << 20)`

I thought of something similar to the way the buttons and switches state is read in the main.c of the gpio_demo, but I am not sure what's the right address for Pmod ports, like for "button right" it's  1 << 19 and for "button up" it's 1 << 20, etc.

According to the pulpino datasheet, the status of the  buttons and switches are read from the INTSTATUS, the interrupt status register, which actually has 32 bits. So I am wondering the bits from 21 to 32 for which ports are they used ? 

any help would be appreciated, thanks a lot

sofchet, 2018-12-16:
@FrancescoConti  any ideas please ?

quangdaovu, 2018-12-17:
First of all, you're misunderstood. There is no "address for Pmod ports", there is only address for GPIO Peripheral. That address is internal to the PULP core, while Pmod is kind of "external", it is just a wire that connect GPIO pins of the core with actual inout pin of FPGA.

As stated in pulpemu_top (https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/fpga/rtl/pulpemu_top.v#L385-L387), GPIO in/out ports of Pulpino core is connected to` gpio_in` & `gpio_out` signal. You need to read, therefore you should take a look at how `gpio_in` is assigned (https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/fpga/rtl/pulpemu_top.v#L225-L228). Note that the pin assignment for `sw_i` and `btn_i` are listed in impl.tcl (https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/fpga/pulpemu/tcl/impl.tcl#L30-L41). Now if you need Pmod, just create another wire named `pmod_i`, assign its value to `gpio_in` and also do pin assignment for it. Afterward you can just use gpio_demo to read.

sofchet, 2018-12-17:
@quangdaovu  thanks a million for your detailed answer, I appreciate. 
So actually I would have last question: 
I realized that one of the Pmods that I need to use is already connected to a wire by default: 

1- From the impl.tcl there is the following line: 

`set_property PACKAGE_PIN Y11  [get_ports {ext_tdi_i}];     # "P-MOD: JA1"`

2- and from the pulpemu_top.v the "ext_tdi_i" signal is assigned to "td_i" wire as follows: 

`assign td_i    = ext_tdi_i;`

and we have as well : 

`.tdi_i             ( td_i           ),`

So my question is: how am I supposed to read this tdi_i signal using a similar core to main.c of gpio_demo ?

quangdaovu, 2018-12-18:
Based on the port name, `ext_tdi_i` is part of JTAG interface, in which you connect the external JTAG cable and debug the PULP core. Of course if you don't need JTAG, you can remove `ext_tdi_i` or move it to another pin, meanwhile assign Y11 according to your need.

Basically it is something like this:

- Move external JTAG to other pins and create new port for GPIO:
From:
https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/fpga/pulpemu/tcl/impl.tcl#L196-L200
to:

>   set_property PACKAGE_PIN <other_pin_here> [get_ports {ext_tdi_i}];     # "P-MOD: JA1"
>   set_property PACKAGE_PIN <other_pin_here> [get_ports {ext_tms_i}];     # "P-MOD: JA2"
>   set_property PACKAGE_PIN <other_pin_here> [get_ports {ext_tck_i}];     # "P-MOD: JA3"
>   set_property PACKAGE_PIN <other_pin_here> [get_ports {ext_tdo_o}];     # "P-MOD: JA4"
>   set_property PACKAGE_PIN <other_pin_here> [get_ports {ext_trstn_i}];   # "P-MOD: JA7"
>   set_property PACKAGE_PIN Y11  [get_ports {JA[0]}];  # "JA1"
>   set_property PACKAGE_PIN AA11 [get_ports {JA[1}];  # "JA2"
>   set_property PACKAGE_PIN Y10  [get_ports {JA[2]}];  # "JA3"
>   set_property PACKAGE_PIN AA9  [get_ports {JA[3]}];  # "JA4"
>   set_property PACKAGE_PIN AB11 [get_ports {JA[4]}];  # "JA7"
>   set_property PACKAGE_PIN AB10 [get_ports {JA[5]}];  # "JA8"
>   set_property PACKAGE_PIN AB9  [get_ports {JA[6]}];  # "JA9"
>   set_property PACKAGE_PIN AA8  [get_ports {JA[7]}];  # "JA10"

Note that the `other_pin_here` part is your job. You need to find the suitable FPGA pin to assign them to (may be other PMOD pins?).

- Declair JA port and connect it to GPIO part of PULP core:
From
https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/fpga/rtl/pulpemu_top.v#L45-L50

to

>  ext_tms_i,
>  ext_tdo_o,
>  JA
>  );
>
>  input  [7:0] JA;
>  inout  [14:0] DDR_addr;
>  inout  [2:0]  DDR_ba;

From
https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/fpga/rtl/pulpemu_top.v#L226
to

>  assign gpio_in[15:8]  = JA;

- Your JA PMOD now connected to `gpio_in[15:8]`, read them using GPIO_Demo.

closed 2018-12-18 by sofchet