pulp-platform/pulpino #269: Add hardware counter

sofchet, 2019-01-02:
Hi, 
I am trying to add a hardware counter to be able to correctly estimate the frequency of external signals. The counter itself is the easiest part, but I am not sure whether I am supposed to connect it to the AXI bus or to the APB? I m not sure as well in which file the counter block should be added and what's the changes to make? 

I would really appreciate any help. 

Many thanks,

sofchet, 2019-01-05:
@FrancescoConti  @gautschimi   any ideas please ?

gautschimi, 2019-01-07:
I'd add a apb slave and connect it on the apb bus. but it might as well be that you can determine the frequency with the gpio module. e.g. by registering interrupts on rising edges of this external signal and then checking the time between the two interrupts.
depends of course how often you need to determine the frequency and how precise it has to be

sofchet, 2019-02-01:
@gautschimi many thanks for your answer, actually I have tried doing this using interrupts, everything goes fine until I reach frequencies higher than 35KHz, at this point the pulpino is simply saturated, and as you mentioned the accuracy as well was not that good. 

That's why I did  think about adding a hardware counter, can you please give me some guidelines about how I can add it as an APB slave ? what are the modification to be made and on which files ?

I really appreciate. thanks