pulp-platform/pulpino #272: A mistake in the document

hyf6661669, 2019-01-21:
I have downloaded this repo, used the command ```make all``` in pulpino/doc/datasheet and got a ```datasheet.pdf```. I think this PDF file is the description of PULPINO SoC, but I think there is a mistake in ```Figure 2.1: PULPINO memory map```: 
[image]

In fact, the reset value of PC is determined by ROM_START_ADDR, which is set to 0x0000_8000 in pulpino/rtl/includes/config.sv:
[image]
Meanwhile, I have found that in pulpino/sw/ref/link.boot.ld, the start address of rom is also set to 0x0000_8000:
[image]

So I think the start address of Boot ROM in the PDF file is wrong, which should be 0x0000_8000 instead of 0x0008_0000.

Besides, I couldn't open the 2 links in pulpino/doc/datasheet/README.MD. They both returned "404 Not Found", they should also be fixed.

AlvinRolling, 2019-04-12:
I think you are right. This has already been recognized by others. If you could read Chinese or use some translation [here](https://cnrv.gitbooks.io/riscv-soc-book/content/ch8/sec2-PULPino_overview.html)

hyf6661669, 2019-04-13:
@AlvinRolling 
Actually I'm Chinese. Anyway thanks for your reply :)

closed 2019-04-13 by hyf6661669

stefanct, 2019-04-16:
Please do not close legitimate bug reports if they are not fixed. This is not a help forum where you should close the topic when you have solved your problem.