pulp-platform/pulpino #274: pulp-riscv-gnu-toolchain build failed

nvlinh87, 2019-01-23:
Hi Pulpino team,

I'm trying to build toolchain for Pulpino by below command:
./configure --prefix=/opt/riscv --with-arch=rv32g --with-abi=ilp32d
make linux

However, it reported fatal error:
td_thr_getfpregs.c: In function 'td_thr_getfpregs':
td_thr_getfpregs.c:44:35: error: dereferencing pointer to incomplete type 'prfpregset_t {aka union __riscv_fp_state}'
     memset (regset, '\0', sizeof (*regset));
                                   ^~~~~~~
../o-iterator.mk:9: recipe for target '/home/linh/tmp/pulp-riscv-gnu-toolchain/build-glibc-linux-32/nptl_db/td_thr_getfpregs.os' failed
make[3]: * [/home/linh/tmp/pulp-riscv-gnu-toolchain/build-glibc-linux-32/nptl_db/td_thr_getfpregs.os] Error 1

I also asked my friend working on an embedded system to build it and he reported the same error.

Please help to fix it.
My system is centos 6.10 with gcc 7.3

Thank you.

closed 2019-01-27 by nvlinh87

AshwiniPohekar, 2019-02-11:
I am also getting the same error. 

Is there any solution available for this ?

nvlinh87, 2019-02-13:
@AshwiniPohekar I switched to use https://github.com/pulp-platform/ri5cy_gnu_toolchain and can compile and run simulation