pulp-platform/pulpino #277: many regression tests failed

nvlinh87, 2019-01-28:
After successfully run helloworld sim (used cmake_configure.riscvfloat.gcc.sh), 
I tried to do regression by below command (from https://github.com/pulp-platform/pulpino/blob/master/sw/README.md)

ctest -L "riscv|sequential" --timeout 1000

However, I got many timeout results:
The following tests FAILED:
         41 - testIRQ.test (Timeout)
         54 - testMacNorm.test (Timeout)
         57 - testDebug.test (Timeout)
         60 - bitDescriptor.test (Timeout)
         67 - sudokusolver.test (Timeout)
         70 - matrixMul8.test (Timeout)
         72 - matrixMul16.test (Timeout)
         76 - convolution.test (Timeout)
         78 - cnn_layer.test (Timeout)
         82 - non_separable_2d_filter.test (Timeout)

Does anyone can help to point out what I need to fix to make them PASS ?
I already increased timeout counter to 1000ns.

Also, I just want to focus on RI5CY core, which test suite I should run and debug ?

bluewww, 2019-03-12:
If you just want to test the RI5CY core we use PULPissimo and PULP testbenches for that. RI5CY also has a stand alone tb that runs riscv-tests and riscv-compliance tests, currently under the `core_tb` branch (merged soon to master).

kingweio, 2019-04-24:
> If you just want to test the RI5CY core we use PULPissimo and PULP testbenches for that. RI5CY also has a stand-alone tb that runs riscv-tests and riscv-compliance tests, currently under the `core_tb` branch (merged soon to master).

I have the same problem. So，what you mean is that the results of  the command” ctest -L "riscv|sequential" --timeout 100“  can only be a timeout? If so, why add this command to the readme? In addition, I would like to ask, what is the meaning of this sequential test? What test is it doing?