pulp-platform/pulpino #278: Where can the "ETH gcc compiler" to be found ?

ogheri, 2019-01-29:
Hi, the documentation speaks of the possibility to build either with the gnu riscv toolchain or the eth compiler...

In fact, I was not able to get anything completed with the gnu toolchain (-march=CAPITAL LETTERS instead of the -march=bla bla bla as expected by the gnu toolchain, -mabi wrong )...

thats why I decided to give a try to the "eth risc v compiler" but never found a github repo for it or anything at all in internet, if for that...

is now obsolete ?

if yes, how about providing cmake configure shell scripts that work for an riscv 32bit gcc gnu toolchain installation perhaps ???

thx!

ogheri, 2019-01-29:
sorry is this one ?

https://github.com/pulp-platform/ri5cy_gnu_toolchain

mentioned in another answer here some time ago ?

FrancescoConti, 2019-01-29:
Yes. The whole software stack for PULPino is legacy, and it does not track current developments of RISC-V GCC. I think you have to use the legacy compiler you linked if you want to run code on PULPino using those scripts and that software stack. Am I right @haugoug ?

That said, RI5CY, Zero-RISCY and Micro-RISCY should support code compiled by the official compiler just fine, and in fact for our more updated platforms (PULPissimo microcontroller https://github.com/pulp-platform/pulpissimo , OPENPULP platform https://github.com/pulp-platform/pulp ) we use a much newer version of the compiler.

haugoug, 2019-01-30:
For pulpino I have no idea, this was done a long time ago by someone else. I guess the best option is to use the latest compiler (https://github.com/pulp-platform/pulp-riscv-gnu-toolchain) and adapt a bit pulpino build system to make it work

ogheri, 2019-01-30:
Hi thx to both of you.
After I downloaded the eth toolchain now that i know where it is, with it at least one of the configure..... sh scripts works enough that i can run the helloworld on questasim.

Surely later perhaps I will fix the scripts to use the latest gnu toolchain...
rgds
Alex

⁣Inviato da BlueMail ​

FrancescoConti, 2019-01-30:
@ogheri in case you do, let us know if you want to contribute that back! :)

closed 2019-01-30 by FrancescoConti