pulp-platform/pulpino #281: Unable to use the cmake_configure.riscv.gcc.sh to build a program, output "is not able to compile a simple test program."

ludblom, 2019-02-22:
The toolchain I am using is "pulp-gnu-toolchain", link it to path in the cmake_configure file and have all the dependencies I need except ModelSim, but will not generate a ModelSim test file so it should not be needed. I want to use a ZedBoard FPGA instead I have. In the fpga folder I read that I should read the README in the "pulpino/sw" folder. Doing that I am guided to make a directory, copy the cmake_configure file, modify it to my needs and run it. However, this is the output I get from it.

`-- The C compiler identification is GNU 7.1.1
-- The CXX compiler identification is GNU 7.1.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/pulp-riscv/bin/riscv32-unknown-elf-gcc
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
Your CMakeCache.txt file was copied to CopyOfCMakeCache.txt. Please send that file to cmake@www.cmake.org.
-- Check for working C compiler: /opt/pulp-riscv/bin//riscv32-unknown-elf-gcc
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Check for working C compiler: /opt/pulp-riscv/bin//riscv32-unknown-elf-gcc -- broken
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "/opt/pulp-riscv/bin//riscv32-unknown-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/ludblom/Desktop/Kandidatarbete/pulpino/sw/build/CMakeFiles/CMakeTmp
    
    Run Build Command:"/usr/bin/gmake" "cmTC_b8c5f/fast"
    /usr/bin/gmake -f CMakeFiles/cmTC_b8c5f.dir/build.make CMakeFiles/cmTC_b8c5f.dir/build
    gmake[1]: Entering directory '/home/ludblom/Desktop/Kandidatarbete/pulpino/sw/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_b8c5f.dir/testCCompiler.c.o
    /opt/pulp-riscv/bin//riscv32-unknown-elf-gcc   -O3 -m32 -g    -o CMakeFiles/cmTC_b8c5f.dir/testCCompiler.c.o   -c /home/ludblom/Desktop/Kandidatarbete/pulpino/sw/build/CMakeFiles/CMakeTmp/testCCompiler.c
    riscv32-unknown-elf-gcc: error: unrecognized command line option '-m32'; did you mean '-mL2='?
    CMakeFiles/cmTC_b8c5f.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_b8c5f.dir/testCCompiler.c.o' failed
    gmake[1]: *** [CMakeFiles/cmTC_b8c5f.dir/testCCompiler.c.o] Error 1
    gmake[1]: Leaving directory '/home/ludblom/Desktop/Kandidatarbete/pulpino/sw/build/CMakeFiles/CMakeTmp'
    Makefile:126: recipe for target 'cmTC_b8c5f/fast' failed
    gmake: *** [cmTC_b8c5f/fast] Error 2
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:20 (enable_language)

-- Configuring incomplete, errors occurred!
`

Khazus, 2019-03-26:
Hi, I have the same issue. Have you fixed it?

ludblom, 2019-03-27:
Yes I have, it seems like it is something wrong with the flags, more specifically the "-m32" flag. To remove it we used this script 

`#!/bin/bash

# Find and replace all occurrances of '-m32' and fix rest of line.

for file in $(find); do
    if [[ -f $file ]]; then
        [[ $(cat $file | grep m32) ]]
        if [[ $? == 0 ]]; then
            echo writing...
            echo $file
           sed 's/\-m32//g' $file > tmp && mv tmp $file
       fi
    fi
done`

You also need to fix your linker file wich was done like this

`#!/bin/bash

# Fixes linker problem by creating default link script.
# Further editing is neccessary!!

riscv32-unknown-elf-ld --verbose | head -n -1 | tail -n +7 | sed '168 a \ \ _fbss = .;' | sed '169 a \ \ . = .;' > /home/path/to/pulpino/sw/build/CMakeFiles/CMakeTmp/riscv.ld`

You also have to change the TARGET_C_FLAGS to "-O3 -march=rv32g -g" instead of what they are now. Good luck!

Khazus, 2019-03-29:
Thank you for your response.

I tried to execute both scripts and i modified TARGET_C_FLAGS. But i have a similar result:

```
-- The C compiler identification is GNU 7.1.1
-- The CXX compiler identification is GNU 7.1.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/riscv/bin/riscv32-unknown-elf-gcc
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Check for working C compiler: /opt/riscv/bin/riscv32-unknown-elf-gcc
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Check for working C compiler: /opt/riscv/bin/riscv32-unknown-elf-gcc -- broken
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "/opt/riscv/bin/riscv32-unknown-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/arcenegui/pulpinoexperiment/pulpino/sw/build/CMakeFiles/CMakeTmp
    
    Run Build Command:"/usr/bin/gmake" "cmTC_d107c/fast"
    /usr/bin/gmake -f CMakeFiles/cmTC_d107c.dir/build.make CMakeFiles/cmTC_d107c.dir/build
    gmake[1]: Entering directory '/home/arcenegui/pulpinoexperiment/pulpino/sw/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_d107c.dir/testCCompiler.c.o
    /opt/riscv/bin/riscv32-unknown-elf-gcc   -O3 -march=rv32g -g    -o CMakeFiles/cmTC_d107c.dir/testCCompiler.c.o   -c /home/arcenegui/pulpinoexperiment/pulpino/sw/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_d107c
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d107c.dir/link.txt --verbose=1
    /opt/riscv/bin/riscv32-unknown-elf-gcc -O3 -march=rv32g -g     CMakeFiles/cmTC_d107c.dir/testCCompiler.c.o  -o cmTC_d107c 
    /opt/riscv/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot open linker script file riscv.ld: No such file or directory
    collect2: error: ld returned 1 exit status
    CMakeFiles/cmTC_d107c.dir/build.make:97: recipe for target 'cmTC_d107c' failed
    gmake[1]: *** [cmTC_d107c] Error 1
    gmake[1]: Leaving directory '/home/arcenegui/pulpinoexperiment/pulpino/sw/build/CMakeFiles/CMakeTmp'
    Makefile:126: recipe for target 'cmTC_d107c/fast' failed
    gmake: *** [cmTC_d107c/fast] Error 2
```
Then, which version of riscv-toolchain are you using? ri5cy-gnu-toolchain or pulp-riscv-gnu-toolchain?

Thank you and regards.

blaklinten, 2019-03-29:
The line containing 'cannot open linker script file riscv.ld: No such file or directory' suggest that the linker script is not to be found. As suggested by @mbolivar in https://github.com/pulp-platform/pulpino/issues/240#issuecomment-443011984 the linker can however provide said script and that is what one of my script does. However; when compiling it would seems as the process entails several test runs and after each test the linker script is deleted. I solved this problem

Khazus, 2019-04-01:
it's work! thank you!

sunil3690, 2019-09-04:
hello,
Thank you. Will you please share the fix-linker script.?
Thanks
PVJ

caifuxi, 2019-10-24:
I ran into the same issue before. Then I changed to [https://github.com/riscveval/ri5cy-gnu-toolchain](url) and problem solved

sunil3690, 2019-10-24:
Hi..
It's working
I configure cmake_configure.riscv.gcc.sh inside the build folder using
ri5cy_gnu_toolchain.
Thanks for your help

sunil3690, 2019-10-24:
Hello

I have facing problem with all application including pulpino platform are
execute using modelsim? During run time we need to add all wave to design?

I run application as testALU using modelsim.
But how to figure out the c code of testALU and simulation waveform please
give the solution.
Thanks

baraujo3007, 2020-06-30:
Hello, I'm facing the same issue, but neither of the response solve it.

That's my cmake_configure.riscv.gcc.sh file

#!/bin/bash

export PATH=/opt/riscv/bin/riscv32-unknown-elf-gcc/:${PATH}

OBJDUMP=`which riscv32-unknown-elf-objdump`
OBJCOPY=`which riscv32-unknown-elf-objcopy`

COMPILER=`/opt/riscv/bin/riscv32-unknown-elf-gcc`
RANLIB=`which riscv32-unknown-elf-ranlib`

#VSIM=`/home/Bruna/intelFPGA_pro/19.4/modelsim_ase/bin/vsim`

TARGET_C_FLAGS="-O3 -march=rv32g -g"
#TARGET_C_FLAGS="-O2 -g -falign-functions=16  -funroll-all-loops"

# if you want to have compressed instructions, set this to 1
RVC=0

# if you are using zero-riscy, set this to 1, otherwise it uses RISCY
USE_ZERO_RISCY=0

# set this to 1 if you are using the Floating Point extensions for riscy only
RISCY_RV32F=0

# zeroriscy with the multiplier
ZERO_RV32M=1
# zeroriscy with only 16 registers
ZERO_RV32E=0

# riscy with PULPextensions, it is assumed you use the ETH GCC Compiler
GCC_MARCH="IMXpulpv2"
#compile arduino lib
ARDUINO_LIB=1

PULP_GIT_DIRECTORY=../../
SIM_DIRECTORY="$PULP_GIT_DIRECTORY/vsim"
#insert here your post-layout netlist if you are using IMPERIO
PL_NETLIST=""

cmake "$PULP_GIT_DIRECTORY"/sw/ \
    -DPULP_MODELSIM_DIRECTORY="$SIM_DIRECTORY" \
    -DCMAKE_C_COMPILER="$COMPILER" \
    -DVSIM="$VSIM" \
    -DRVC="$RVC" \
    -DRISCY_RV32F="$RISCY_RV32F" \
    -DUSE_ZERO_RISCY="$USE_ZERO_RISCY" \
    -DZERO_RV32M="$ZERO_RV32M" \
    -DZERO_RV32E="$ZERO_RV32E" \
    -DGCC_MARCH="$GCC_MARCH" \
    -DARDUINO_LIB="$ARDUINO_LIB" \
    -DPL_NETLIST="$PL_NETLIST" \
    -DCMAKE_C_FLAGS="$TARGET_C_FLAGS" \
    -DCMAKE_OBJCOPY="$OBJCOPY" \
    -DCMAKE_OBJDUMP="$OBJDUMP"

# Add -G "Ninja" to the cmake call above to use ninja instead of make

And what I get when I try to run it:

riscv32-unknown-elf-gcc: fatal error: no input files
compilation terminated.
-- The C compiler identification is unknown
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/c++
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
CMake Error at CMakeLists.txt:20 (enable_language):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Check for working CXX compiler: /usr/bin/c++
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Detecting CXX compile features - done
-- Configuring incomplete, errors occurred!
See also "/home/Bruna/pulpino/sw/build/CMakeFiles/CMakeOutput.log".
See also "/home/Bruna/pulpino/sw/build/CMakeFiles/CMakeError.log".

What do I have to do to solve it?
Thanks

Unlimitosu, 2024-02-14:
@ludblom I ran the shell script that removes -m32 flags then I fixed `TARGET_C_FLAGS` into `-O3 -march=rv32g -g`.

After that, I ran `cmake_configure.riscv.gcc.sh` and I encountered the same errors: `  The C compiler "/opt/risqv/bin/riscv32-unknown-elf-gcc" is not able to compile a simple test program.`

What is the linker script you are saying? I cannot find `pulpino/sw/build/CMakeFiles/CMakeTmp/riscv.ld`. I didn't success to run any cmake_* script. 

Can you help me about this?

ludblom, 2024-03-13:
@Unlimitosu Have you tried @blaklinten 's solution?

Unlimitosu, 2024-03-13:
@ludblom I don't know what is fix-linker script... and I think it's because I used WSL2.
I tried again in pure Ubuntu 20.04,  then I didn't see that error, but there's something error with path. 
When I run `make vcompile`, error said  .csh file(maybe) did not exist. and I didn't solved it yet.
Because it is early morning here, I will reply here in this weekend.

Thanks!

shane1604, 2024-05-12:
[image]
Hi Im facing thiss error. Anyone has solution for this pls help :<