pulp-platform/pulpino #282: Unable to open file `riscv_defines.pvk'  Synthesize dc_compiler error

sofchet, 2019-02-23:
Hi, 
I am trying to synthesize the pulpino using Synopsys dc_compiler, I have a script that was able to successfully synthesize other circuits, but it is failing for the pulpino RTL files. I am getting this error over and over again while running the synthesize: 

**` Unable to open file `riscv_defines.pvk**': in search_path {/gpfs/data/backup2/template . /gpfs/runtime/opt/synopsys/H-2013.03-SP2/libraries/syn /gpfs/runtime/opt/synopsys/H-2013.03-SP2/minpower/syn /gpfs/runtime/opt/synopsys/H-2013.03-SP2/dw/syn_ver /gpfs/runtime/opt/synopsys/H-2013.03-SP2/dw/sim_ver}. (VER-292) Error:  /users/sh33/data/backup2/rtl/riscv_ex_stage.sv:38: Package 'riscv_defines' has not been analyzed for import or content extraction. (VER-224)` 

any ideas about this  Unable to open file riscv_defines.pvk error ? 

many thanks

virinchiroy, 2020-10-25:
Include WORK directory in your search path. The error should go away.

QuanXChen, 2020-11-25:
Hi, I have the same problem as you. But I can’t find work directory。 I can't find the   riscv_defines.pvk file, how can I do to solve this problem?