pulp-platform/pulpino #289: What is the clock frequency of riscy core on zedboard?

jurevreca12, 2019-04-04:
Hello,
I synthesized pulpino for the zedboard, and it works fine. I am trying to figure out the clock at which the core is running. I got different information depending on where I look:
      - The top constains file for the fpga says the clock should be 50ns or 20MHz
      - The spiload program sets the CSR register and says the default clock is 5MHz
      - I modified the pulpemo_top.v file so that it outputs the input clock to the pulpino system to      an LED. The resulting clock was 1.666 MHz. But I am not sure if they change the clock inside the pulpino. Before it reaches the core.

Is there any documentation about the clock configuration in pulpino? Or, how can I now at which clock frequency the Riscy core is running?

Thank you for your help,
Jure

PS. I am NOT using the zero-riscy core.

hyf6661669, 2019-04-07:
@jurevreca12 
Hi, take a look at 
pulpino/fpga/ips/xilinx_clock_manager/tcl/run.tcl:
```
set_property -dict [list CONFIG.INTERFACE_SELECTION {Enable_AXI} CONFIG.USE_DYN_RECONFIG {true}
 CONFIG.PRIM_IN_FREQ {50.000} CONFIG.PRIMARY_PORT {clk50_i} CONFIG.CLK_OUT1_PORT {clk_o} 
CONFIG.CLKOUT1_REQUESTED_OUT_FREQ {5.000} CONFIG.LOCKED_PORT {rst_no} CONFIG.RESET_TYPE 
{ACTIVE_LOW} 
CONFIG.RESET_PORT {rst_ni}] [get_ips xilinx_clock_manager]
```
The port CLK_OUT1 is used for core clock. Therefore, according to the .tcl file, the default clock for pulpino on ZedBoard is 5MHz, which seems very slow. 

Can you try to use different frequency to find the max frequency that pulpino can run normally on ZedBoard? Since at present I don't have a Vivado license, it will be very nice of you to help me. I 'm tring to put pulpino on zedboard with FPU.

jurevreca12, 2019-04-08:
@hyf6661669 
I believe you are able to synthesize with the free Vivado WebPack. But anyway that may very well be the clock in. But it is not neccesairly the clock of the core, which is the clock in which I am interested.
I will try to increase the clock as much as possible, but first I have to know exactly how much is it.

sunil3690, 2019-10-17:
> Hello,

> I synthesized pulpino for the zedboard, and it works fine. I am trying to figure out the clock at which the core is running. I got different information depending on where I look:
> - The top constains file for the fpga says the clock should be 50ns or 20MHz
> - The spiload program sets the CSR register and says the default clock is 5MHz
> - I modified the pulpemo_top.v file so that it outputs the input clock (s_clk_pulpino) to the pulpino system to an LED. The resulting clock was 1.666 MHz. But I am not sure if they change the clock inside the pulpino. Before it reaches the core.
> 
> Is there any documentation about the clock configuration in pulpino? Or, how can I now at which clock frequency the Riscy core is running?
> 
> Thank you for your help,
> Jure
> 
> PS. I am NOT using the zero-riscy core.

how to synthesized pulpino on zedboard. i am unable to set envirnment variable to select the board