pulp-platform/pulpino #292: Problem while running "cmake_configure.riscv.gcc.sh" file

Mehrdad6991, 2019-04-10:
when I execute" ./cmake_configure.riscv.gcc.sh" inside the build folder I receive this:

-- The C compiler identification is unknown
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
Your CMakeCache.txt file was copied to CopyOfCMakeCache.txt. Please send that file to cmake@www.cmake.org.
CMake Error at CMakeLists.txt:20 (enable_language):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/mehrdad/My_files/git/pulpino/sw/CMakeFiles/CMakeOutput.log".
See also "/home/mehrdad/My_files/git/pulpino/sw/CMakeFiles/CMakeError.log".

Mehrdad6991, 2019-04-10:
I also installed the toolchain of : https://github.com/pulp-platform/ri5cy_gnu_toolchain

but I receive this:

In file included from /home/mehrdad/My_files/git/pulpino/sw/ri5cy_gnu_toolchain-master/build/src/newlib-gcc/gcc/cp/except.c:1023:0:
cfns.gperf: In function ‘const char* libc_name_p(const char*, unsigned int)’:
cfns.gperf:101:1: error: ‘const char* libc_name_p(const char*, unsigned int)’ redeclared inline with ‘gnu_inline’ attribute
cfns.gperf:26:14: note: ‘const char* libc_name_p(const char*, unsigned int)’ previously declared here
cfns.gperf: At global scope:
cfns.gperf:26:14: warning: inline function ‘const char* libc_name_p(const char*, unsigned int)’ used but never defined
Makefile:1064: recipe for target 'cp/except.o' failed
make[4]: *** [cp/except.o] Error 1
make[4]: Leaving directory '/home/mehrdad/My_files/git/pulpino/sw/ri5cy_gnu_toolchain-master/build/build-gcc-newlib/gcc'
Makefile:4112: recipe for target 'all-gcc' failed
make[3]: *** [all-gcc] Error 2
make[3]: Leaving directory '/home/mehrdad/My_files/git/pulpino/sw/ri5cy_gnu_toolchain-master/build/build-gcc-newlib'
Makefile:867: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/mehrdad/My_files/git/pulpino/sw/ri5cy_gnu_toolchain-master/build/build-gcc-newlib'
Makefile:214: recipe for target 'stamps/build-gcc-newlib' failed
make[1]: *** [stamps/build-gcc-newlib] Error 2
make[1]: Leaving directory '/home/mehrdad/My_files/git/pulpino/sw/ri5cy_gnu_toolchain-master/build'
Makefile:28: recipe for target 'build' failed
make: *** [build] Error 2

Douglasas, 2019-04-18:
I had this error as well, to fix this I changed the function declaration in the file **ri5cy_gnu_toolchain/build/src/newlib-gcc/gcc/cp/cfns.h**. Apparently it's identifying the function declaration twice, so I commented the line 57, that only declares the function name and parameters. I  did successfully run the code on the zedboard.
Below there is the line that I commented.
`//const char * libc_name_p (const char *, unsigned int);`

Mehrdad6991, 2019-04-23:
> I had this error as well, to fix this I changed the function declaration in the file **ri5cy_gnu_toolchain/build/src/newlib-gcc/gcc/cp/cfns.h**. Apparently it's identifying the function declaration [...]

Thanks. without changing any file by changing my OS to ubuntu 16.4 and start from the beginning by installing riscy toolchain that problem solved. now I face another problem that after using minicom I am not able to boot!

closed 2019-05-06 by Mehrdad6991