pulp-platform/pulpino #304: error when running update-ips.py

nglphucvinh, 2019-05-23:
Hello,

I got this error when running the ./update-ips.py for the first time after cloning the repo

```python
Using remote git server https://github.com, remote is https://github.com/pulp-platform
Cloning into 'ipstools'...
remote: Enumerating objects: 21, done.
remote: Counting objects: 100% (21/21), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 813 (delta 9), reused 16 (delta 8), pack-reused 792
Receiving objects: 100% (813/813), 159.19 KiB | 329.00 KiB/s, done.
Resolving deltas: 100% (561/561), done.
Traceback (most recent call last):
  File "./update-ips.py", line 91, in <module>
    ipdb.update_ips(remote = remote)
TypeError: update_ips() got an unexpected keyword argument 'remote'
```
How can I fix this error? Thanks in advacne.

tfiorucci, 2019-05-23:
I do have the same error, two days ago i had a running pulpino installed on another machine so i was perfectly able to run `./update-ips.py`
today i have the same errors. I something has been changed probably.

eactor, 2019-05-28:
I have the same problem. It comes from 

https://github.com/pulp-platform/IPApproX/commits/verilator/ipstools/IPDatabase.py 

If you use the old version with:
git checkout 8f0e144757a639cf9133296cc8e7425c49d9b53b
this works for now.

tfiorucci, 2019-05-28:
thanks for your solution, but the tree reference seems not to exist...

```shell
[tfiorucci@localhost pulpino]$ git checkout 8f0e144757a639cf9133296cc8e7425c49d9b53b
fatal: reference is not a tree: 8f0e144757a639cf9133296cc8e7425c49d9b53b
```

am i doing something wrong or the reference is misspelled?

eactor, 2019-05-29:
you have to change into the ipstools folder, this folder is created by update_ips.py.

nglphucvinh, 2019-05-29:
I have tried your solution, but the same error still occurs

Checkout in the ipstools folder:
```shell
bill@bill:~/pulpino/ipstools$ git checkout 8f0e144757a639cf9133296cc8e7425c49d9b53b
HEAD is now at 8f0e144 Added default site-dependent path
```
Comment out some lines in update-ips.py to import the ipstools
```python
# download IPApproX tools in ./ipstools and import them
if os.path.exists("ipstools") and os.path.isdir("ipstools"):
    # cwd = os.getcwd()
    # os.chdir("ipstools")
    # execute("git pull origin verilator", silent=True)
    # os.chdir(cwd)
    import ipstools
else:
    # try to find the ipstools repository
    if "http" in remote:
        if execute("git clone %s/IPApproX.git ipstools -b verilator" % (remote)) != 0:
            execute("git clone %s/pulp-tools/IPApproX.git ipstools -b verilator" % (server))
    else:
        if execute("git clone %s/IPApproX.git ipstools -b verilator" % (remote)) != 0:
            execute("git clone %s:pulp-tools/IPApproX.git ipstools -b verilator" % (server))

    import ipstools
``` 
Run update-ips.py and the problem still persist:
```shell
bill@bill:~/pulpino$ ./update-ips.py 
Using remote git server https://github.com, remote is https://github.com/pulp-platform
Traceback (most recent call last):
  File "./update-ips.py", line 91, in <module>
    ipdb.update_ips(remote = remote)
TypeError: update_ips() got an unexpected keyword argument 'remote'

```

eactor, 2019-05-29:
Sorry for confusion, the problem should be fixed in general my workaround is only a dirty workaround:

> I have tried to change line 91 in update-ips.py into
> `ipdb.update_ips(remote)`
> but then it gives me permission errors:

In addition I changed platform/IPApproX/commits/verilator/ipstools/IPDatabase.py at line 100:
from
`default_server="git@github.com",`
to 
`default_server="https://github.com",`

Not sure if that has any thing to do with the problem. 
During that process a git pop-up came up asking for my credentials, after that I got an email telling me: [GitHub] A personal access token has been added to your account. ....

FrancescoConti, 2019-05-29:
My fault. I did not know the "verilator" branch was used in PULPino, and I updated it upstream... Will fix in the next few minutes.

closed 2019-05-29 by FrancescoConti

FrancescoConti, 2019-05-29:
Should be working now, please re-open the issue if it doesn't.

koolca, 2019-05-29:
IPs updated successfully!
Branch 'verilator' set up to track remote branch 'verilator' from 'origin'.
Switched to a new branch 'verilator'
Traceback (most recent call last):
  File "./generate-scripts.py", line 46, in <module>
    ipdb.export_vsim(script_path="vsim/vcompile/ips", target_tech='umc65')
AttributeError: 'IPDatabase' object has no attribute 'export_vsim'

nglphucvinh, 2019-05-29:
> Should be working now, please re-open the issue if it doesn't.

Thanks! `./update_ips.py` is working properly now, the next problem is in `./generate-script.py`, I think you should also replace the `verilator` -> `verilator-pulpino` for it to work properly.

FrancescoConti, 2019-05-29:
> > Should be working now, please re-open the issue if it doesn't.
> 
> Thanks! `./update_ips.py` is working properly now, the next problem is in `./generate-script.py`, I think you should also replace the `verilator` -> `verilator-pulpino` for it to work properly.

you're right...

FrancescoConti, 2019-05-29:
> IPs updated successfully!
> Branch 'verilator' set up to track remote branch 'verilator' from 'origin'.
> Switched to a new branch 'verilator'
> Traceback (most recent call last):
> File "./generate-scripts.py", line 46, in 
> ipdb.export_vsim(script_path="vsim/vcompile/ips", target_tech='umc65')
> AttributeError: 'IPDatabase' object has no attribute 'export_vsim'

You should remove the `ipstools` (which will now be pointing at a much more recent version than what works with PULPino) and `git pull` in the PULPino root before using `update-ips` again.

koolca, 2019-05-29:
> You should remove the `ipstools` (which will now be pointing at a much more recent version than what works with PULPino) and `git pull` in the PULPino root before using `update-ips` again.

Thanks！

commit ab45bbde (2019-05-29): Update update-ips.py

Fixes #304
commit c1b78849 (2019-05-29): Previous commit did *not* fix #304, this one should...
commit 6631c49b (2019-05-29): Fixes #304 also for generate-scripts