pulp-platform/pulpino #309: Improving Modelsim Simulation Performance of Pulpino

alp010, 2019-06-28:
Hi,
I am using klessydra core ([on](https://github.com/klessydra/T03x)) pulpino and I can run my tb from modelsim. Problem is simulation takes a lot of time (like 12 hours for running helloworld from the core). Klessydra is vhdl and pulpino is verilog. Is problem can be this (I don't think so)
) or it is about student edition of modelsim? I want to carry my simulation environment to vivado for improving performance. Do you have any suggestion for run simulations from vivado or improving my simulations performance, I try several things but can't solve yet.

Thanks,

babekmemmedov1985, 2020-09-17:
Hi, I try to implement pulpino but I couldn't could you please help me and tell me how to do it in ModelSim because I couldn't run it. Or could you tell me which software Do I need to download

alp010, 2020-09-17:
I get modelsim from İntel page. 
It's been a long time, but it was like this:https://fpgasoftware.intel.com/?product=modelsim_ae#tabs-2
Also my modelsim path in ubuntu is this: /home/.../intelFPGA_pro/18.1/modelsim_ase/linuxaloem

I had many problems with pulpino on Ubuntu. There may be places in pulpino where you need to remove the 64 bit flags. I can't help with that because I don't remember exactly. 
Good luck!

babekmemmedov1985, 2020-09-18:
Thank you very much