pulp-platform/pulpino #323: Memory Error

Ankasysemre, 2019-09-30:
Hi,
Technology memory  instantiated in Pulpino riscv instate of sp_ram model but instantiated memory works for instruction memory not work for data memory . Why Am ı getting this error ? I can read and write to this technology memory model correctly.
Thaks for your help.

aignacio, 2019-10-23:
Pls describe a bit better your issues on writing to the memory, are you referring to **contention** issues? I mean, AMBA standard follows the concept of ADDRESS and DATA phases split into clock moments, sometimes if the memory cannot handle this _pipelinish_ model, you need to avoid misaligned req. or writing and reading in seq. to the same address.

sunil3690, 2019-10-23:
Thanks for your reply.
I will try