pulp-platform/pulpino #324: Running fpga application(led_demo)  c code using ri5cy-gnu-toolchain.

sunil3690, 2019-10-21:
hi,
i am running application c code(led_demo) on pulpino using ri5cy-gnu toolchain.
a new folder, in
which the C program will be copied, has to be created in the path sw/apps. The files are
managed by using CMake that exploits the CMakeLists.txt files to organize the applica-
tion folders. In each of these a CMakeLists.txt has to be created and structured like the
following example: my c code folder name **led_demo  main.c**
add_application(led_demo main.c)
in which led_demo is the name used when we run the application and main.c is the
name of the program. 
Also the CMakeLists.txt in the sw/apps has to be modified including the line add_subdirectory,

i am facing an error while executing "make main" comand

root@reva-Extensa-M200-H110:~/pulpino/sw/build_demo/led_demo# make main
cc     main.c   -o main
main.c:12:24: fatal error: string_lib.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'main' failed
make: *** [main] Error 1

please can anyone help the issue
thaks in advance

aignacio, 2019-10-21:
I guess you're not including [this](https://github.com/pulp-platform/pulpino/tree/master/sw/libs/string_lib/inc) dir inside your toolchain rules... (pls check -Isw/libs/string_lib/inc)

sunil3690, 2019-10-21:
Thank you very much for your reply
I will try it

sunil3690, 2019-10-22:
Thanks you so much it's working.

aignacio, 2019-10-22:
pls consider close the issue

closed 2019-10-22 by sunil3690

sunil3690, 2019-10-22:
hi
i am running pulpino on FPGA board selected as Zybo not externally
connected.

set the environment variable BOARD. In the tcsh script file as steven BOARD
"Zybo"
also i did not specify the BOAD PART and BOARD NUMBER it will take default
XILINX_PART "xc7z020clg484-1" XILINX_BOARD "em.avnet.com:zynq:zed:c"
inside the environment variable set "RISCY_RV32F"

and set the vivado toolchain PATH and Xilinx SDK PATH as

export PATH=/root/pulpino/ri5cy_gnu_toolchain/install/bin:$PATH
export LD_LIBRARY_PATH=/root/pulpino/Xilinx/Vivado/2019.1/lib/lnx64.o
export PATH=/root/pulpino/Xilinx/Vivado/2019.1/bin:$PATH
export PATH=/root/pulpino/Xilinx/SDK/2019.1/bin:$PATH
export PATH=/root/pulpino/Xilinx/SDK/2019.1/gnu/microblaze/lin/bin:$PATH
export
PATH=/root/pulpino/Xilinx/SDK/2019.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:$PATH
export
PATH=/root/pulpino/Xilinx/SDK/2019.1/gnu/microblaze/linux_toolchain/lin64_le/bin:$PATH
export PATH=/root/pulpino/Xilinx/DocNav:$PATH

After this i type "make all" targated to *"cmake_configure.riscvfloat.gcc.sh
<http://cmake_configure.riscvfloat.gcc.sh>"* in the fpga directory i am
getting an error as
root@sunil-Inspiron-3542:~/pulpino/fpga# make all
make -C ips/xilinx_fp_fma/ clean all
make[1]: Entering directory '/root/pulpino/fpga/ips/xilinx_fp_fma'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_fp_fma.xpr
rm -rf xilinx_fp_fma.hw
rm -rf xilinx_fp_fma.data
rm -rf xilinx_fp_fma.runs
rm -rf xilinx_fp_fma.srcs
rm -rf xilinx_fp_fma.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
vivado -mode batch -source tcl/run.tcl

****** Vivado v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
  **** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
# if { ![info exists ::env(XILINX_PART)] } {
#   set ::env(XILINX_PART) "xc7z020clg484-1"
# }
# if { ![info exists ::env(XILINX_BOARD)] } {
#   set ::env(XILINX_BOARD) "em.avnet.com:zynq:zed:c"
# }
# set partNumber $::env(XILINX_PART)
# set boardName  $::env(XILINX_BOARD)
# create_project xilinx_fp_fma . -part $partNumber
ERROR: [Common 17-53] User Exception: Project already exists on disk,
please use '-force' option to overwrite:
     /root/pulpino/fpga/ips/xilinx_fp_fma/xilinx_fp_fma.ip_user_files
INFO: [Common 17-206] Exiting Vivado at Mon Oct 21 16:41:39 2019...
Makefile:4: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/root/pulpino/fpga/ips/xilinx_fp_fma'
Makefile:33: recipe for target 'ips/xilinx_fp_fma/ip/xilinx_fp_fma.dcp'
failed
make: *** [ips/xilinx_fp_fma/ip/xilinx_fp_fma.dcp] Error 2

please let me know the issue.
Thank you
also please let know how to Prepare the SD card and the ZedBoard for
booting via SD card and Extract the content of the rootfs.tar archive and
put it on the second partition of the SD card

OR

it will enough to direct run the
"Running applications on PULPino"
please can help the issues
Thank you

sunil3690, 2019-11-22:
i am unable to rectifying the following error please help issues
please mention the steps regarding the solution

U-Boot 2015.01-dirty (Nov 13 2019 - 09:18:31)

Board:  Xilinx Zynq
I2C:   ready
DRAM:  ECC disabled 512 MiB
MMC:   zynq_sdhci: 0
SF: Detected S25FL256S_64K with page size 256 Bytes, erase size 64 KiB,
total 32 MiB
In:    serial
Out:   serial
Err:   serial
Board:  Xilinx Zynq
Net:   Gem.e000b000
Error: Gem.e000b000 address not set.

Hit any key to stop autoboot:  0
Device: zynq_sdhci
Manufacturer ID: 27
OEM: 5048
Name: SD04G
Tran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 3.7 GiB
Bus Width: 4-bit
reading uEnv.txt
** Unable to read file uEnv.txt **
Copying Linux from SD to RAM...
reading uImage
3479944 bytes read in 335 ms (9.9 MiB/s)
reading devicetree.dtb
10264 bytes read in 16 ms (626 KiB/s)
## Booting kernel from Legacy Image at 02080000 ...
   Image Name:   Linux-3.18.0-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3479880 Bytes = 3.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Kernel Image ... OK
   Loading Device Tree to 1ed1e000, end 1ed23817 ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 3.18.0-xilinx (root@sunil-Inspiron-3542) (gcc version 4.9.1
(Sourcery CodeBench Lite 2014.11-30) ) #1 SMP PREEMPT Wed Nov 13 09:9
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: xlnx,zynq-7000
cma: Reserved 16 MiB at 0x1f000000
Memory policy: Data cache writealloc
PERCPU: Embedded 10 pages/cpu @5efd1000 s8768 r8192 d24000 u40960
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
Kernel command line: console=ttyPS1,115200 ip=on root=/dev/mmcblk0p2 rw
earlyprintk
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 496192K/524288K available (4695K kernel code, 241K rwdata, 1620K
rodata, 216K init, 219K bss, 28096K reserved, 0K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[... 156 lines cut ...]
EXT3-fs (mmcblk0p2): error: couldn't mount because of unsupported optional
features (240)
EXT2-fs (mmcblk0p2): error: couldn't mount because of unsupported optional
features (240)
EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
VFS: Mounted root (ext4 filesystem) on device 179:2.
devtmpfs: mounted
Freeing unused kernel memory: 216K (40634000 - 4066a000)
EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
Starting logging: OK
Initializing random number generator... random: dd urandom read with 5 bits
of entropy available
done.
Starting network...
ip: RTNETLINK answers: File exists
Starting sshd: OK

Welcome to Buildroot
buildroot login: root
#
#

I am not able to login using buildroot
Thanks
Sunil