pulp-platform/pulpino #326: Running application on pulpino (Vivado webpack 2019.1) zynq fpga board

sunil3690, 2019-10-22:
hi
i am running pulpino on FPGA board selected as Zybo not externally connected.

set the environment variable BOARD. In the tcsh script file as steven BOARD "Zybo"
also i did not specify the BOAD PART and BOARD NUMBER it will take default
XILINX_PART "xc7z020clg484-1" XILINX_BOARD "em.avnet.com:zynq:zed:c"
inside the environment variable set "RISCY_RV32F"

and set the vivado toolchain PATH and Xilinx SDK PATH as

export PATH=/root/pulpino/ri5cy_gnu_toolchain/install/bin:$PATH
export LD_LIBRARY_PATH=/root/pulpino/Xilinx/Vivado/2019.1/lib/lnx64.o
export PATH=/root/pulpino/Xilinx/Vivado/2019.1/bin:$PATH
export PATH=/root/pulpino/Xilinx/SDK/2019.1/bin:$PATH
export PATH=/root/pulpino/Xilinx/SDK/2019.1/gnu/microblaze/lin/bin:$PATH
export PATH=/root/pulpino/Xilinx/SDK/2019.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:$PATH
export PATH=/root/pulpino/Xilinx/SDK/2019.1/gnu/microblaze/linux_toolchain/lin64_le/bin:$PATH
export PATH=/root/pulpino/Xilinx/DocNav:$PATH

After this i type "make all" targated to "cmake_configure.riscvfloat.gcc.sh" in the fpga directory i am getting an error as
root@sunil-Inspiron-3542:~/pulpino/fpga# make all
make -C ips/xilinx_fp_fma/ clean all
make[1]: Entering directory '/root/pulpino/fpga/ips/xilinx_fp_fma'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_fp_fma.xpr
rm -rf xilinx_fp_fma.hw
rm -rf xilinx_fp_fma.data
rm -rf xilinx_fp_fma.runs
rm -rf xilinx_fp_fma.srcs
rm -rf xilinx_fp_fma.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
vivado -mode batch -source tcl/run.tcl

****** Vivado v2019.1 (64-bit)
**** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
**** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
if { ![info exists ::env(XILINX_PART)] } {
set ::env(XILINX_PART) "xc7z020clg484-1"
}
if { ![info exists ::env(XILINX_BOARD)] } {
set ::env(XILINX_BOARD) "em.avnet.com:zynq:zed:c"
}
set partNumber $::env(XILINX_PART)
set boardName $::env(XILINX_BOARD)
create_project xilinx_fp_fma . -part $partNumber

ERROR: [Common 17-53] User Exception: Project already exists on disk, please use '-force' option to overwrite:
/root/pulpino/fpga/ips/xilinx_fp_fma/xilinx_fp_fma.ip_user_files
INFO: [Common 17-206] Exiting Vivado at Mon Oct 21 16:41:39 2019...
Makefile:4: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/root/pulpino/fpga/ips/xilinx_fp_fma'
Makefile:33: recipe for target 'ips/xilinx_fp_fma/ip/xilinx_fp_fma.dcp' failed
make: *** [ips/xilinx_fp_fma/ip/xilinx_fp_fma.dcp] Error 2

please let me know the issue.
Thank you
also please let know how to Prepare the SD card and the ZedBoard for booting via SD card and Extract the content of the rootfs.tar archive and put it on the second partition of the SD card

OR

it will enough to direct run the
"Running applications on PULPino"
please let me know
Thank you

aignacio, 2019-10-23:
You're trying to recreate the vivado project that had already been started, please modify the [building scripts](https://github.com/pulp-platform/pulpino/blob/master/fpga/pulpino/tcl/run.tcl#L59) with _-force_ option like I do [here](https://github.com/aignacio/riscv_verilator_model/blob/master/fpga/main/tcl/start.tcl#L77) =)

sunil3690, 2019-10-25:
i am not getting this how to modify the scripts as you mentioned above
please address me clearly
i am getting same error.
otherwise need to connect an external fpga boards?
please let me know
thank you
make: *** [clean] Error 2
root@reva-Extensa-M200-H110:~/pulpino/fpga# make all
make -C ips/xilinx_clock_manager/ clean all
make[1]: Entering directory '/root/pulpino/fpga/ips/xilinx_clock_manager'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_clock_manager.xpr
rm -rf xilinx_clock_manager.hw
rm -rf xilinx_clock_manager.data
rm -rf xilinx_clock_manager.runs
rm -rf xilinx_clock_manager.srcs
rm -rf xilinx_clock_manager.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
rm -rf .Xil
vivado -mode batch -source tcl/run.tcl

****** Vivado v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
  **** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
# if { ![info exists ::env(XILINX_PART)] } {
#   set ::env(XILINX_PART) "xc7z020clg484-1"
# }
# if { ![info exists ::env(XILINX_BOARD)] } {
#   set ::env(XILINX_BOARD) "em.avnet.com:zynq:zed:c"
# }
# set partNumber $::env(XILINX_PART)
# set boardName  $::env(XILINX_BOARD)
# set ila_name xilinx_clock_manager
# create_project $ila_name . -part $partNumber
ERROR: [Common 17-53] User Exception: Project already exists on disk,
please use '-force' option to overwrite:

 /root/pulpino/fpga/ips/xilinx_clock_manager/xilinx_clock_manager.ip_user_files
INFO: [Common 17-206] Exiting Vivado at Fri Oct 25 10:02:19 2019...
Makefile:4: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/root/pulpino/fpga/ips/xilinx_clock_manager'
Makefile:13: recipe for target
'ips/xilinx_clock_manager/ip/xilinx_clock_manager.dcp' failed
make: *** [ips/xilinx_clock_manager/ip/xilinx_clock_manager.dcp] Error 2
root@reva-Extensa-M200-H110:~/pulpino/fpga#

aignacio, 2019-10-25:
I guess the error "**ERROR: [Common 17-53] User Exception: Project already exists on disk,**" talk by itself

sunil3690, 2019-10-25:
Hi
Sry for intrupt

Now it's working

I remember some missing steps

Thanks for the help

sunil3690, 2019-10-25:
Previously I will use vivado webpack 2019.1 version which the pulpino
platform is not supported.and also it will not able to access path as well
as updated files present in recent vivado toolchain.
The pulpino is tested by using vivado 2015.1 version.
Naw I am using vivado 2015.1 version
Naow it's fine with testing by zedboard

My question is that wthout connecting zedboard it is possible to get
generate an bitstream using vivado toolchain?
Please let me know
Thank you so much.😃

aignacio, 2019-10-27:
yes, for sure, bitstream gen. does have anything related to the board itself....pls close the issue

closed 2019-10-27 by sunil3690

sunil3690, 2019-11-14:
Hi,
How fpga application are run on pulpino core and generate the bitstream of
each application.
Please help the issue.
Thank you.

mo2men3la2, 2022-12-09:
@sunil3690 
Hi, can you tell me how you got Vivado 2015.1 version as I also have 2019.1 version which is not compatible with PULPino platform.

eactor, 2022-12-09:
Under xilinx.com/download is menu entry called Vivado Archive 

https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vivado-design-tools/archive.html