pulp-platform/pulpino #342: Cannot find module definition for L2_MB_INTERCO

pgraykowski, 2020-04-06:
I'm working to get Pulpino to compile in Cadence Xcellium.    I've hit an issue where there is a missing definition for module L2_MB_INTERCO

This module is instanced/parameterized in two files, however none of the source files have this module definition.

xmelab: *E,CUVMUR (**../ips/axi/axi_mem_if_DP/axi_mem_if_MP_Hybrid_multi_bank.sv,582**|15): instance 'axi_mem_if_MP_Hybrid_multi_bank.L2_MB_INTERCO' of design unit 'XBAR_TCDM_FC' is unresolved in 'worklib.axi_mem_if_MP_Hybrid_multi_bank:sv'.
   L2_MB_INTERCO
               |
xmelab: *E,CUVMUR (../ips/axi/axi_mem_if_DP/axi_mem_if_multi_bank.sv,546|15): instance 'axi_mem_if_DP_multi_bank.L2_MB_INTERCO' of design unit 'XBAR_TCDM_FC' is unresolved in 'worklib.axi_mem_if_DP_multi_bank:sv'.

Is this file missing from the repository?

cuixuyang615, 2022-02-10:
> I'm working to get Pulpino to compile in Cadence Xcellium. I've hit an issue where there is a missing definition for module XBAR_TCDM_FC / L2_MB_INTERCO
> 
> code snippet:
> 
> XBAR_TCDM_FC #(...) L2_MB_INTERCO ( ... )
> 
> This module is instanced/parameterized in two files, however none of the source files have this module definition.
> 
> xmelab: *E,CUVMUR (**../ips/axi/axi_mem_if_DP/axi_mem_if_MP_Hybrid_multi_bank.sv,582**|15): instance 'axi_mem_if_MP_Hybrid_multi_bank.L2_MB_INTERCO' of design unit 'XBAR_TCDM_FC' is unresolved in 'worklib.axi_mem_if_MP_Hybrid_multi_bank:sv'. L2_MB_INTERCO | xmelab: *E,CUVMUR (../ips/axi/axi_mem_if_DP/axi_mem_if_multi_bank.sv,546|15): instance 'axi_mem_if_DP_multi_bank.L2_MB_INTERCO' of design unit 'XBAR_TCDM_FC' is unresolved in 'worklib.axi_mem_if_DP_multi_bank:sv'.
> 
> Is this file missing from the repository?

I excluded it from src_files.yml. It seems that module has not been instantiated in top files.

darphee1025, 2024-12-27:
I have the same issue , could u find the XBAR_TCDM_FC interface .

darphee1025, 2024-12-27:
fixed this issue , u can remove this relation file . cause useless