pulp-platform/pulpino #343: Illegal Instruction (core 0) at PC 0x00000084

Jaina-96, 2020-05-04:
Hello, I'm using a custom 4k SRAM in replacement to the 32k SRAM, and I get the error as seen below in the ModelSim simulation when executing a simple HelloWorld program. I have updated the ram size in all the RTL files. The link.common.ld file has not been updated as I am not expanding the instruction/data ram. Thanks in advance.
[image]

barrydebruin, 2020-05-26:
You need to update the riscv.common.ld file to account for the reduced size instruction memory. The stack will be written to the end of the memory, which has changed for you.

nguyendaithien, 2023-09-29:
i had the same problem, how did you fix it ?? Thank you