pulp-platform/pulpino #344: pulpino floating point unit

Ankasysemre, 2020-05-05:
Hi,
After fpu is up in pulpino. I can not use it from software part without inline assembly. 
Anyone can help How can I do for example multupication operation like "3.4*2.43" in c software part.
Thanks.

sunil3690, 2020-05-05:
Are you sure about to enable the fpu unit in pulpino platform.(RISCY core)

Ankasysemre, 2020-05-05:
Hi sunil3690 ,
Sorry there is wrong word in up . I use it with inline assembly in simulation enviroment. My main question How can I use it like "3.4*2.43". I suppose that something goes wrong with c compilization part. For example may be there is some flags to generate instruction with fpu support during compilizaiton c source code.

sunil3690, 2020-05-05:
Pulpino support only integer multiplication (RV32IM) IMXPulpv2. It might be
wrong with your c code, you needed to write simple two integer
multiplication c code and run on it will be on right way if successfully
executed. But as per your requirement two(2.4*3.4)floating point
multiplication is not supported. It will not executed with fpu. For that we
need to configure for fpu then only it is possible. Present it is
configured only for RV32IM. Not for RV32F=1. According to my study. Because
I am also trying to go for fpu(RISCY)

barrydebruin, 2020-05-26:
To add; its maybe convenient to error on all implicit double promotions i.e. add the following
-fsingle-precision-constant -mfpdouble=float" -Werror=double-promotion
to your TARGET_C_FLAGS variable.

The RISC-V core doesn't support any double instructions.

closed 2020-07-24 by Ankasysemre