pulp-platform/pulpino #347: update-ips.py command doesn't work

baraujo3007, 2020-06-24:
I'm new at pulp and i'm trying to build pulpino, i have build the riscv-toolchain.  When i try to clone the repositories of the version control using ./update-ips.py I get this:

Using remote git server https://github.com, remote is https://github.com/pulp-platform
From https://github.com/pulp-platform/IPApproX
 * branch            verilator-pulpino -> FETCH_HEAD
Traceback (most recent call last):
  File "./update-ips.py", line 76, in <module>
    import ipstools
  File "/home/Bruna/pulpino/ipstools/__init__.py", line 15, in <module>
    from .IPDatabase import *
  File "/home/Bruna/pulpino/ipstools/IPDatabase.py", line 16, in <module>
    import yaml
ImportError: No module named yaml

What am i supposed to do?

vedanjalip, 2020-06-25:
Try running this on the terminal

pip install pyyaml

sunil3690, 2020-06-25:
Update your python yaml first.

baraujo3007, 2020-06-29:
> Try running this on the terminal pip install pyyaml
> […](#)

I did it, but it still doesn't work. the same error appears.

closed 2020-06-30 by baraujo3007

phuong2403, 2020-10-01:
I have the same problem. I tried to do: pip3 install pyyaml
Then the result was: Requirement already satisfied: pyyaml in /usr/lib64/python3.6/site-packages
Then I did ./update-ips.py, but it still doesnt work.
Please anyone can help?