pulp-platform/pulpino #360: ./update-ips.py doesnt work. ImportError: No module named yaml

phuong2403, 2020-10-01:
+ I was trying to run update sub-repositories  by using: **python2.7 ./update-ips.py**  (only **./update-ips.py** didn't work with me)
Then I got as following:
---------------------------------------------------------------------------------
Using remote git server https://github.com, remote is https://github.com/pulp-platform                                              From https://github.com/pulp-platform/IPApproX                     * branch            verilator-pulpino -> FETCH_HEAD              Traceback (most recent call last):                                  File "./update-ips.py", line 76, in <module>                        import ipstools                                                 File "/volume1/scratch/tnguyen/pulpino/ipstools/__init__.py", line 15, in <module>                                                    from .IPDatabase import *                                       File "/volume1/scratch/tnguyen/pulpino/ipstools/IPDatabase.py", line 16, in <module>                                                  import yaml                                                   ImportError: No module named yaml
---------------------------------------------------------------------------------
+ I tried to follow the issue #347:

**pip3 install pyyaml**

Then the result was: Requirement already satisfied: pyyaml in /usr/lib64/python3.6/site-packages and it is pyyaml 3.12

+ Then I did **python2.7 ./update-ips.py** again, but it still doesn't work.

Please anyone can help?

closed 2020-10-02 by phuong2403

mohsinshahbaz-10xe, 2021-04-29:
@phuong2403 How did you resolve the issue?

phuong2403, 2021-04-29:
> @phuong2403 How did you resolve the issue?

@mohsinshahbaz-10xe  I install yaml package of python2

mohsinshahbaz-10xe, 2021-04-29:
@phuong2403 Thanks, worked!