pulp-platform/pulpino #370: make vcompile error

satabdi25-2000, 2021-02-19:
vlib: Command not found.
vlib: Command not found.

--> Compiling PULPino Platform...

--> Compiling PULP IPs libraries...
--> Compiling axi_node...
vlib: Command not found.
vmap: Command not found.
Compiling component:  axi_node

vlog: Command not found.

make[3]: *** [CMakeFiles/vcompile.dir/build.make:57: CMakeFiles/vcompile] Error 1
make[2]: *** [CMakeFiles/Makefile2:3601: CMakeFiles/vcompile.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:3608: CMakeFiles/vcompile.dir/rule] Error 2
make: *** [Makefile:155: vcompile] Error 2

Got this when doing the make vcompile in the pulpino/sw/build folder. kindly help if any solutions to this.

Felix-ziteng, 2022-12-22:
Maybe you need to check your Questasim version

JacoboJin, 2023-12-28:
Maybe you need to check your Modelsim/Questasim version(if you have installed it).
And if you dont, you should be supposed to install Modelsim/Questasim and ensure it work properly.

haresh-seenivasagan, 2024-01-08:
Hi @Felix-ziteng and @JacoboJin what would be the recommended version. 
I am using  Model Technology ModelSim ALTERA STARTER EDITION vsim 10.5b Simulator 2016.10 Oct  5 2016
but am facing the same issue
Ubuntu 18.04